OpenSprinkler/OpenSprinkler-Firmware #190: Stations are not resetting after restart

ilouzl, 2021-09-15:
In a case where a program is running and activates any station, if there was a server reset (such as ```sudo /etc/init.d/OpenSprinkler.sh restart```), the station is stay active even though the program is not running anymore, so it will never reset automatically (unless other program resets it sometime).
I think there should be a force reset of all the station upon initialization.

rayshobby, 2021-09-15:
The OpenSprinkler::begin() function already does this:
https://github.com/OpenSprinkler/OpenSprinkler-Firmware/blob/master/OpenSprinkler.cpp#L801

ilouzl, 2021-09-16:
That's weird because it won't close the stations in my setup.
My stations are configured as GPIO's with active low.
Also, after reset, the active stations won't stop if using the manual 'stop all stations' button.
Any idea on that?

rayshobby, 2021-09-16:
I think I may have an explanation: on Arduino static variables are initialized to 0 at reboot; but this isn't a standard behavior on C/C++ as static variables need to be explicitly initialized. This would affect station_bits array,

ilouzl, 2021-09-16:
I’m on raspberry pi. Does it matter?

ilouzl, 2021-09-19:
> The OpenSprinkler::begin() function already does this:
> https://github.com/OpenSprinkler/OpenSprinkler-Firmware/blob/master/OpenSprinkler.cpp#L801

Is it possible that this reset is called before the station attributes was red from the nvm?
As far as I understand, the station data is loaded here (just after the reset):
https://github.com/OpenSprinkler/OpenSprinkler-Firmware/blob/3c70d3b838c4471f303aed8f56844c61a6047651/main.cpp#L364

rayshobby, 2021-09-19:
nvm has nothing to do with station_bits -- nvm stores non-volatile data (options etc.), station_bits is volatile.
The issue has to do with this is a GPIO (i.e. a special station) and not a standard zone -- at reboot if it doesn't see a change in status, it won't send an explicit turn-off command to special zones. This can be easily fixed by explicitly turn off special zones at reboot.

ilouzl, 2021-09-19:
How do I do that?

rayshobby, 2022-12-16:
It can be done by adding a loop for example here:
https://github.com/OpenSprinkler/OpenSprinkler-Firmware/blob/master/OpenSprinkler.cpp#L807
that loops over all stations and call switch_special_station(sid, 0). Specifically:

`for(byte sid=0;sid<nstations;sid++) switch_special_station(sid, 0);`

I will add this to firmware 2.2.0 code.

closed 2022-12-16 by rayshobby