OpenSprinkler/OpenSprinkler-Firmware #191: OpenSprinkler does not close connection normally, but sends back an RST (TCP reset flag)

gerdneuman, 2021-09-22:
I use a reverse proxy (apache) in front of OSPI using a config like listed in https://opensprinkler.com/forums/topic/mobile-app-unable-to-connect-through-ssl-reverse-proxy-apache/#post-40021

I see lots of these error message in the apache log (like 1 new error every two seconds) while surfing the opensprinkler web app:
```
[Tue Sep 21 08:39:13.280067 2021] [proxy_http:error] [pid 22540:tid 2911843360] (104)Connection reset by peer: [client 91.66.126.XX:44360] AH01110: error reading response, referer: https://mydomain.com:30283/opensprinkler/
```

Searching the forum I found this bug report: https://opensprinkler.com/forums/topic/openspinkler-server-sends-back-rst-using-httptiny/ Also searching the web for the `Connection reset by peer` error, it indicates that the OSPI does not seem to close the connection gracefully.

rayshobby, 2022-11-03:
Which firmware version is this referring to?

gerdneuman, 2022-11-04:
That's with Version `2.1.9 (7)` on OSPi. 

Albeit the error message I do not see any effects by this.

rayshobby, 2022-11-04:
I am not sure how to address this issue to be honest. It likely has to do with the firmware calling
m_client->stop();
after finishing sending the last packet. But I am not sure how to address this.

closed 2022-11-04 by rayshobby