OpenSprinkler/OpenSprinkler-Firmware #193: 64 Byte MQTT usernames needed

MinimalPTH, 2021-11-09:
I am using an MQTT broker called flespi. Authentication is handled using a 64 byte user name. Example:

2bypoemSP5tE9gf0O4jRu7nMQZMlVsSNtg3FjdnY8Fq6JvWnn35H0ghG4TPPbxrs

More info: https://flespi.com/kb/tokens-access-keys-to-flespi-platform

I am unable to use this username as it is longer than the field in the OpenGarage UI will allow me to paste.

Can the username field accommodate a username long enough to handle these strings?

rayshobby, 2022-02-13:
It seems this question is for OpenGarage, not OpenSprinkler.

closed 2022-02-13 by rayshobby

sumoneelse, 2022-02-13:
It affects both OpenSprinkler and OpenGarage

rayshobby, 2022-02-13:
You can use the OpenSprinkler API to set MQTT name. You don't need to be constrained by the UI.