OpenSprinkler/OpenSprinkler-Firmware #195: [Feature Request] Add support for ESP-01 controlled relay modules.

GregFrost, 2021-12-27:
I have been able to get the OpenSprinkler firmware (with some modifications) to work on an ESP-01 controlled relay board like the ones shown at these link:
[AliExpress Link](https://www.aliexpress.com/item/32963978306.html)
[Amazon Link](https://www.amazon.com.au/DollaTek-ESP8266-Channel-Cellphone-Control/dp/B07HC7SJK1/ref=sr_1_20?crid=1IJE89XNOFULH)

The trick with these boards is that the relay is controlled by sending a 4 byte binary code on the serial line 115200/8/n/1 from the ESP module (there is another non-reprogrammable MCU on the board that listens to the serial commands and controls the relays). e.g to turn on relay 1 on you send the 4 byte hex code a0, 01, 01, a2. To turn it off you send a0, 01, 00, a1 (so it looks like the first byte a0 is a delimiter, then the second byte is the relay number, the third is on or off, and the last byte is a checksum).

I have also found that I need to add a 20msec delay between controlling different relays for them to reliably switch.

There are many other boards like this, with differing numbers of relays (I have seen single channel relays and 2 channel relay boards).

With these boards, all you likely need to add is a full bridge rectifier with a buck converter to power the board from the solenoid 24V relay line, and you have a very inexpensive headless WIFI sprinkler controller (no display or buttons, you need to rely on the phone/browser interface).

I would love it if support for this type of board could be added to OpenSprinkler. I got mine working by hacking the switching logic for the RF station type in OpenSprinkler::switch_rfstation to send the appropriate serial command. To do it properly, you would have to add a new station type (perhaps named Serial Binary, or something like that), where you specified the binary sequence to send to turn the relay on and the sequence to turn it off. That would require updates to the ui code which is out of my league (hence I only co-opted the RF station type in my copy of the code).

Is this something that you would consider adding?

PS: the forum seems to be broken, as that may have been a more appropriate place to post this, but whenever I am logged in, the forum links give me Page Not Found error, preventing me from being able to post.

GregFrost, 2021-12-30:
See it working here: https://www.reddit.com/r/functionalprint/comments/rrtfod/inexpensive_wifi_sprinkler_controller/

GregFrost, 2022-01-02:
PS: the hack I did to get this working is in this fork:
https://github.com/GregFrost/OpenSprinkler-Firmware

garudaonekh, 2022-01-20:
Great! Have you tried this with 8 relay board?
By the way, how stable is it?

I never tried with ESP-01, I have used ESP32. I keep the ESP32 running for more than 3 weeks without any problem(However, i don't attached any relay to it yet, just the board).

garudaonekh, 2022-01-20:
Can you clarify the part on the Sprinkler App to make it work with your ESP01 board? Which station type to use?

GregFrost, 2022-01-20:
> Can you clarify the part on the Sprinkler App to make it work with your ESP01 board? 
See the fork I posted.

Which station type to use?
RF with the code set to the relay number (single digit). You also need to set the station delay to 5 seconds to space out the commands to the relays.

garudaonekh, 2022-01-21:
ok,how many PINs left? Can this work with ESP01 8 relay?

GregFrost, 2022-01-21:
> ok,how many PINs left? Can this work with ESP01 8 relay?

The relays switch based on sequences sent on the serial connection, so if the 8 relay board works the same way (which it probably does?) then it would still work with the esp-01.

amiseh, 2022-06-17:
Dear GregFrost,

during whole day I'm trying to make Yout fork working with my LC Technology 4 channel relay board. unfortunately without any success :(

can You tell me what should I put in RF station type to make it working??

on tasmota I have to configure it like below to allow tamosta to work becuase without the belows configuration it is not turning on and off the relays on the LC board... 

can You tell me where You have done any changes/this hack to make LC board (with esp8266 module) work with serial connection??

on System#Boot do Backlog Baudrate 115200; SerialSend5 0 endon
  on Power1#State=1 do SerialSend5 A00101A2 endon
  on Power1#State=0 do SerialSend5 A00100A1 endon
  on Power2#State=1 do SerialSend5 A00201A3 endon
  on Power2#State=0 do SerialSend5 A00200A2 endon
  on Power3#State=1 do SerialSend5 A00301A4 endon
  on Power3#State=0 do SerialSend5 A00300A3 endon
  on Power4#State=1 do SerialSend5 A00401A5 endon
  on Power4#State=0 do SerialSend5 A00400A4 endon

THANK YOU in advance for any feedback, help and suggestions.

GregFrost, 2022-06-19:
> Dear GregFrost,
> 
> during whole day I'm trying to make Your fork working with my LC Technology 4 channel relay board. unfortunately without any success :( it doesn't work for me :(
> 
> can You tell me what should I put in RF station type to make it working??

If you have burnt my fork to the ESP-01, then when configuring the RF I’d for the station, set it to the single digit relay number that you want to switch

> on tasmota I have to configure it like below to allow tamosta to work becuase without the belows configuration it is not turning on and off the relays on the LC board...
> 
> can You tell me where You have done any changes/this hack to make LC board (with esp8266 module) work with serial connection??

You need to use my fork of OpenSprinkler which has adapted the RF station type to control the LC board relay.

amiseh, 2022-06-19:
@GregFrost THX a lot for Your fork and Your "hack" in switch_rfstation function. just have found everything and right now everything works as expected. 

THX a lot!

rayshobby, 2022-11-04:
Thank you for creating the fork and make the firmware work with the relay module. I am not sure whether we should include this feature in the official firmware, because the changes are quite specific to the particular 4-channel relay module. I am closing this request for now. Your fork can serve as a reference point for anyone who wants to use the same relay module.

By the way, I took a look at your fork and noticed you mentioned SPIFFS and ESP8266 core 3.0.2. While preparing OpenSprinkler firmware 2.2.0, I encountered a similar issue and therefore have to change to use LittleFS. SPIFFS is deprecated in core 3.0.2 so moving forwards LittleFS would be the standard file system.

closed 2022-11-04 by rayshobby

mondjef, 2025-02-17:
I know this is an old issue but I am pulling at straws atm...

I have an ESP8266 Relay X2 by LC Technology exactly as described in the following link that works as expected with Tasmota with the configuration as described in the link. [https://templates.blakadder.com/LC-Relay-ESP01-2R-5V.html](url)  Based on how this relay board operates it should work with @GregFrost fork I believe.

I would like to use this with OpenSprinkler and have followed the instructions as laid out by @GregFrost by cloning his fork, installing necessary libraries in the Arduino IDE etc and compiling.  I have used all the exact same versions referenced and it compiles just fine and uploads to the ESP8266.  I get the access point and configure the wifi credentials and reboot...all is fine and I can access the OpenSprinkler web gui.  I configure the stations as RF and put the relay # as the RF code but nothing happens when I test the stations.

I have tried to see what is being sent to attempt to verify the communication using the Serial Monitor in the Arduino IDE but I am not really accustomed to this stuff and not sure if I am doing things right....as far as I can see there is nothing being communicated.  Does anybody have any suggestions on how I can trouble shoot this or that could walk me through things I might be able to look at to get this working?  Would be much appreciated.

rayshobby, 2025-02-18:
Honestly you should file this issue to the GregFrost fork. Supporting ESP-01 is currently not a feature that our firmware supports, nor do we intend to support it in the future. So posting this in his fork would be the best way to seek help.

mondjef, 2025-02-23:
> Honestly you should file this issue to the GregFrost fork. Supporting ESP-01 is currently not a feature that our firmware supports, nor do we intend to support it in the future. So posting this in his fork would be the best way to seek help.

I would have but it seems that repository does not allow creating issues.