OpenSprinkler/OpenSprinkler-Firmware #196: WiringPi is now deprecated

sachaw, 2022-01-14:
the WiringPi library now deprecated, and an alternative should be used: http://wiringpi.com/wiringpi-deprecated/
Are there any temporary workarounds for new installs?

sachaw, 2022-02-13:
Can you please reopen this, wiringpi has indeed been removed from recent versions of raspberry pi os, and not installable via apt.
There are also a number of other libraries available that are worth exploring.

rayshobby, 2022-02-13:
OK, I will keep this open. I suppose you are referring to raspbian bullseye which is the newest version of raspbian. I believe wiringpi is still installable via api on buster. So unless if you must use bullseye, just use buster.

roben, 2022-04-11:
This really is an issue. Even though Debian Buster is a LTS release, it will eventually be deprecated. Enforcing old software to support a library abandoned by its own creator creates a huge technical debt and it will just get worse in time.

master5597, 2022-04-19:
It looks like some people picked it up.  Not saying it shouldn't be replaced, but this may help in the interim:[https://github.com/WiringPi/WiringPi](https://github.com/WiringPi/WiringPi)

mathesst, 2022-05-10:
Hi there,

now the original wiringPi Website is completely gone. The Library is also not anymore listed in the debian package search [debian package search](https://packages.debian.org/index). I tried to install the mirror version which went fine, but i am not sure if the normal compilation did take it up. As @roben pointed out, the build script doesn't even stop compiling Opensprinkler in case the apt install went wrong. 

With this stated is it currently even possible to install a new OsPI?

BG Mathias

sachaw, 2022-05-10:
Agreed, WiringPi isn't viable any ore and an alternative needs to be sought.

opensprinklershop, 2022-05-20:
WiringPi is here available: http://wiringpi.com/download-and-install/
has anyone tested it with bullseye?

varac, 2022-06-29:
Ugh, I found this because I followed the [official installation instructions](https://openthings.freshdesk.com/support/solutions/articles/5000631599-installing-and-updating-the-unified-firmware). I installed default RaspiOS (bullseye, 32-bit), but then the install script couldn't install the wireingpi package.

* Please add a warning to the README.md that this project is not maintained anyomore (and point ppl to possible alternatives)
* Please add a warning to the README.md that you need to use the old version of Raspi OS (buster) (or point to any option how to add wiringpi to bullseye)
* I don't know how to address in inclusion of these warnings in the [official installation instructions](https://openthings.freshdesk.com/support/solutions/articles/5000631599-installing-and-updating-the-unified-firmware), but if anyone has a close connection to it, please update them as well.

It's not a great user experience to land in this deprecation hole.

Thanks anyway for this project and I hope maintenance will continue some day!

varac, 2022-06-29:
Btw, this is how I got it installed it on RPi OS bullseye just now:

    wget https://github.com/WiringPi/WiringPi/releases/download/2.61-1/wiringpi-2.61-1-armhf.deb
    sudo dpkg -i wiringpi-2.61-1-armhf.deb

rayshobby, 2022-11-02:
I found that raspi-gpio command is now available in the latest raspbian and that can replace wiringPi for turning on pull-up/pull-down of any GPIO. This will be integrated into the next version of the firmware.

roben, 2022-11-03:
Awesome, thanks!

rayshobby, 2022-11-04:
This fix has been integrated into 2.1.9(11). I tested it on my RPi 0 W, which seems to work. I don't have any newer RPis for testing due to the ongoing shortage, but I am hopeful that it should work for all RPis running Bullseye.

closed 2022-11-04 by rayshobby