OpenSprinkler/OpenSprinkler-Firmware #201: Flow sensor phantom counts

n3xus1, 2022-02-27:
opensprinkler Pi with raspberry pi 2B version 1.1 

using default install of bullseye causes the flow sensor to count with not input connected
issue is not present on default install of legacy buster with same configuration

mattriney, 2022-04-01:
I have this issue as well. There also seems to be an issue with the rain sensor not changing properly when triggered/cleared.

mattriney, 2022-04-01:
Just confirmed rolling back to buster does not show the same phantom flow sensor counts.

mattriney, 2022-04-02:
The flow sensor reading goes erratic after being triggered for the first time and never goes back to baseline as seen in this graph
[image]
.

rayshobby, 2022-11-03:
The issue with bullseye is that it no longer supports wiringPi, which is needed (specifically, the 'gpio' command that comes with wiringPi) to enable the internal pull-up resistor on the sensor pins. This will be resolved in the next firmware by changing to use 'raspi-gpio' command instead.

closed 2022-11-03 by rayshobby