OpenSprinkler/OpenSprinkler-Firmware #202: Unexpected event - "External IP updated"

Nu11u5, 2022-04-08:
For a few days now OpenSprinkler has been logging an event "External IP updated: xxx.xxx.xxx.xxx".  Since I have IFTTT webhooks set up I also get this as a notification.  In the 4 years I've had the OpenSprinkler set up I did not receive this type of event - now it is every few hours.  What's more, **the listed IP is never my public IP**, and for a while the event listed the "127.255.255.255" loopback address.

What is triggering this event?

What is used to determine the external IP?

CoupeWare, 2022-04-28:
I'm having the same problem.

salbahra, 2022-04-28:
Is the weather service working on the device?

I ask because I believe the weather service is what provides the external IP to the device.

CoupeWare, 2022-04-29:
Weather adjustment is set to manual. I just cleared the location since I'm not using the weather adjustment. Weather has not worked well since WU stopped working so I installed a rain sensor.

zhatt, 2022-04-30:
I am seeing this too.  It looks like the info coming back from the weather server is wrong.  Here an example of pulling the info via curl and eip keeps changing.

```
11:33 $ curl 'http://weather.opensprinkler.com/0?loc=42.36,-71.06&wto="key":""&fwv=219'
&tz=32&sunrise=342&sunset=1183&eip=2918528204&rawData={"wp":"Manual"}&errCode=0✔ ~/SRC/opensprinkler/git-OpenSprinkler-Firmware [master|✚ 4…3] 
11:33 $ curl 'http://weather.opensprinkler.com/0?loc=42.36,-71.06&wto="key":""&fwv=219'
&tz=32&sunrise=342&sunset=1183&eip=1822612243&rawData={"wp":"Manual"}&errCode=0✔ ~/SRC/opensprinkler/git-OpenSprinkler-Firmware [master|✚ 4…3] 
11:34 $ curl 'http://weather.opensprinkler.com/0?loc=42.36,-71.06&wto="key":""&fwv=219'
&tz=32&sunrise=342&sunset=1183&eip=1822612293&rawData={"wp":"Manual"}&errCode=0✔ ~/SRC/opensprinkler/git-OpenSprinkler-Firmware [master|✚ 4…3] 
11:34 $ curl 'http://weather.opensprinkler.com/0?loc=42.36,-71.06&wto="key":""&fwv=219'
&tz=32&sunrise=342&sunset=1183&eip=2728279690&rawData={"wp":"Manual"}&errCode=0✔ ~/SRC/opensprinkler/git-OpenSprinkler-Firmware [master|✚ 4…3] 
```

salbahra, 2022-04-30:
The content after `errCode=0` does not appear for me and I believe that's a bad paste somewhere.

The eip field is returning a CloudFlare registered IP so yes it appears to be returning the wrong external IP. I will try to look into this when I can.

zhatt, 2022-04-30:
The content after the errCode was part of my prompt because a newline was missing.  I cleaned it up in my previous comment.

Thanks for looking in to this.

CoupeWare, 2022-05-01:
Clearing the location has eliminated the "External IP Updated" emails. I was not using the Weather Adjustment so I would not have guessed simply having the location set would caused this error. I hope this data point helps determine the source of the problem.

KB4MDD, 2022-05-16:
I am having this problem also.

ssancese, 2022-06-13:
Same issue here.
Weather adjustment is ON and the weather service is working correctly.

Nu11u5, 2022-07-09:
The issue seemed to have stopped for several days without changing anything in the OpenSprinkler config.  However I got several events again as recent as Wednesday (every 2 hours, all day), but none since.

prophead67, 2022-07-13:
Having the same issue. It seemed to stop for a short while, then started back up again. It all started several weeks ago, spamming my phone with IFTTT messages.

I do have location set and use weather adjustment. That function is working as far as I can tell. Just updated to the latest firmware and that didn't make it stop.

Nu11u5, 2022-08-07:
This seems to have stopped being an issue, at least for myself.  Can anyone confirm?

ssancese, 2022-08-08:
Issue still open for me

Nu11u5, 2022-08-08:
Scratch that.

I updated the firmware from 2.1.9rev2 to 2.1.9rev9 and the issue came back.

ssancese, 2022-08-13:
Rev. 2.1.9(9)

Main.cpp line 1017-1026:

		byte wuf = os.weather_update_flag;
		if(wuf) {
			if((wuf&WEATHER_UPDATE_EIP) | (wuf&WEATHER_UPDATE_WL)) {
				// at the moment, we only send notification if water level or external IP changed
				// the other changes, such as sunrise, sunset changes are ignored for notification
				push_message(NOTIFY_WEATHER_UPDATE, (wuf&WEATHER_UPDATE_EIP)?os.nvdata.external_ip:0, (wuf&WEATHER_UPDATE_WL)?os.iopts[IOPT_WATER_PERCENTAGE]:-1);
			}
			os.weather_update_flag = 0;
		}

The issue should be fixed by changing the above to:

		if( os.weather_update_flag & WEATHER_UPDATE_WL)
		{
			// at the moment, we only send notification if water level IP changed
			// the other changes, such as sunrise, sunset changes are ignored for notification
			push_message(NOTIFY_WEATHER_UPDATE, 0, os.iopts[IOPT_WATER_PERCENTAGE]);
			os.weather_update_flag = 0;
		}

The patch compiles OK but unfortunately I am not able to test it for the next few weeks.
Please review the code and if OK create a pull request for Ray.

rayshobby, 2022-12-16:
The PR has been merged. See my comment there:
https://github.com/OpenSprinkler/OpenSprinkler-Firmware/pull/207#issuecomment-1353919539

closed 2022-12-16 by rayshobby