OpenSprinkler/OpenSprinkler-Firmware #205: error: reference to ‘byte’ is ambiguous

cdubbs85, 2022-07-19:
Trying to run the build script on a raspberry pi 3 b+ with fresh install of ubuntu server 22.04 returns quite a few errors. build-essential installs completes successfully. I think the root of the failure has to do with 'byte' type. Any ideas for resolving this?

I think this is the first error:
`OpenSprinkler.h:70:9: error: reference to ‘byte’ is ambiguous`

cdubbs85, 2022-07-19:
Build seems to work fine using Raspberry Pi OS.

luizribeiro, 2022-08-13:
You can build with `-std=c++14`. You are most likely using GCC11 which defaults to `-std=c++17`

kdccisit, 2022-09-25:
I can confirm the build issue with Ubuntu 22.04. I modified the build.sh g++ lines and appended `-std=c++14` to lines 17, 22, and 28. The build completed and the OpenSprinkler service seems to be working properly.

sprinkmeier, 2022-10-18:
I would suggest changing _byte_ to _uint8_t_
https://en.wikipedia.org/wiki/C_data_types#stdint.h
https://cplusplus.com/reference/cstdint/

rayshobby, 2022-12-16:
`-std=c++14` flag has been added to build.sh. I am closing this issue for now. I agree changing byte to uint8_t is better and we will revisit this some time in the future.

closed 2022-12-16 by rayshobby

sprinkmeier, 2023-01-03:
[byte_to_uint8_t.patch.txt](https://github.com/OpenSprinkler/OpenSprinkler-Firmware/files/10334043/byte_to_uint8_t.patch.txt)
patch for your consideration.

amnesiemaniac, 2023-07-23:
Please re-open this issue, it is not fixed.

@rayshobby did not fixed build.sh as he said.
This ist from current version 
```
if [ "$1" == "demo" ]; then
        echo "Installing required libraries..."
        apt-get install -y libmosquitto-dev
        echo "Compiling firmware..."
        g++ -o OpenSprinkler -DDEMO -std=c++14 -m32 main.cpp OpenSprinkler.cpp program.cpp opensprinkler_server.cpp utils.cpp weather.>
elif [ "$1" == "osbo" ]; then
        echo "Installing required libraries..."
        apt-get install -y libmosquitto-dev
        echo "Compiling firmware..."
        g++ -o OpenSprinkler -DOSBO main.cpp OpenSprinkler.cpp program.cpp opensprinkler_server.cpp utils.cpp weather.cpp gpio.cpp eth>
else
        echo "Installing required libraries..."
        apt-get update
        apt-get install -y libmosquitto-dev
        apt-get install -y raspi-gpio
        if ! command -v raspi-gpio &> /dev/null
        then
                echo "Command raspi-gpio is required and is not installed"
                exit 0
        fi
        echo "Compiling firmware..."
        g++ -o OpenSprinkler -DOSPI main.cpp OpenSprinkler.cpp program.cpp opensprinkler_server.cpp utils.cpp weather.cpp gpio.cpp eth>
fi

The option "-std=C++14" is only set in if-case demo not for any others. Build failed.
After changing 
` g++ -o OpenSprinkler -DOSPI **-std=c++14** main.cpp OpenSprinkler.cpp program.cpp opensprinkler_server.cpp utils.cpp weather.cpp gpio.cpp eth>`
now it works

rayshobby, 2023-07-23:
This has been fixed in this commit:
https://github.com/OpenSprinkler/OpenSprinkler-Firmware/commit/d9baa8f715f45fb3765176f449ec71c4ab60bb40
PLEASE, can you report on what platform are you not able to compile? Just saying the 'issue is not fixed' does NOT help. You need to let us know which platform it's failing, so we can fix it.

amnesiemaniac, 2023-07-23:
As I said in my first Comment: Your Fix is not complete. 

Please look at the snippet above, it's right from your build.sh-Script. There is a if-Statement:

If "demo", ...
Else if "osbo" ...
else ...

and your fix just works in first part (if "demo"). There ist no Flag -std=c++14 in both else blocks.
If I try to build a OSPI-Version from your fixed version, it failed.

Thanks!

rayshobby, 2023-07-24:
You said, I quote: "Please re-open this issue, it is not fixed."
You did NOT say: "Your Fix is not complete.".