OpenSprinkler/OpenSprinkler-Firmware #206: [Feature Request] Sleep the hardware display (OLED develops severe burn-in)

Nu11u5, 2022-08-07:
I have the OpenSprinkler hardware operating for a few years now, and the OLED display has severe burn-in and is unreadable.  It may need to be replaced if I am ever required to read information from it.

Please update the firmware to sleep the OLED display until a button is pressed.  This can be an optional setting in the UI.

rayshobby, 2022-08-07:
This is already in the current firmware, by setting Idle Brightness to 0.

closed 2022-08-07 by rayshobby

Nu11u5, 2022-08-07:
@rayshobby where in the UI is this?  I am unable to find it.

rayshobby, 2022-08-07:
Edit Options -> LCD Screen -> Idle Brightness.

Nu11u5, 2022-08-07:
Does this require a minimum firmware version?  My hardware is still running v2.1.9 (2).

[image]

rayshobby, 2022-08-07:
I said 'current firmware', which means 2.1.9(9)

Nu11u5, 2022-08-07:
Thanks I will attempt an update.

Are the release notes for 2.1.9 (9) on the release page cumulative since the last release?  The is not mention of previous releases since 2.1.8.

Nu11u5, 2022-08-07:
I've completed the update to v2.1.9-r9 and set the idle brightness to the lowest value in the UI.  The OLED is dimming but is still lit.  Is this sufficient to prevent further burn-in?