OpenSprinkler/OpenSprinkler-Firmware #213: Local UI host Request

PipeDeveloper, 2022-11-09:
Hello @rayshobby 

According to host the local UI (for raspbery pi). The link on the documetation seems to be broken
https://openthings.freshdesk.com/support/solutions/articles/5000164006-using-a-different-server-for-ui-assets

Also it would be nice to update those files, since the latest files have the option to configure MQTT on the web UI. (Maybe do a repository for them?)

rayshobby, 2022-11-09:
First of all, this is not a firmware issue, it's related to UI, so you can post this in the OpenSprinkler-App repo.

Second, the instructions in that article is no longer necessary. There is a simpler approach, which is to download the OpenSprinkler-App repository, unzip it, and run index.html inside is directly. There is no need to change the script url anymore.

closed 2022-11-09 by rayshobby

PipeDeveloper, 2022-11-09:
Thank you for the quick respone. I'll try it.