OpenSprinkler/OpenSprinkler-Firmware #218: [Issue]Devices Drop from ethernet

KasKam, 2023-01-07:
Hello... We are having an issue with some of our Open sprinkler device where they lose ethernet connection and have to be rebooted to reestablish  the connection.  We have 13 devices on our network and most do not do this. We do have 5 that seem to lose connection as described. We have tried to ruled out all network issues that we could test for. It Almost like they go to sleep and stop taking request usually when no feeds are running. The devices do continue to feed but we have to unplug the devices to get them to reestablish the ethernet connection. We see the same type of issue no mater what patch we are using on the devices. (2.1.10,2.1.11, 2.2.0). Any suggestions or things we could do to see why this is occurring. 

[image]

rayshobby, 2023-01-07:
Are all of them on wired Ethernet connection or some of them are on WiFi?

KasKam, 2023-01-07:
All are on Ethernet.

rayshobby, 2023-01-07:
Well, honestly, I don't know. I know some users have been experiencing issues with wired Ethernet, but we've never been able to reproduce the problem internally, so there is no way for us to debug. The firmware does have a feature where you can use a program to periodically reboot the controller automatically, say every 24 hours or every xx hours. My other suggestion is that if you have WiFi, you may put some of the controllers on WiFi to see if it improves the connectivity. Beyond that, I really don't know what else there is.

KasKam, 2023-01-07:
Is there any way that we can debug and possible capture the issue? It happens quite often on the devices its occurring.

rayshobby, 2023-01-07:
Sorry, I really don't know how. Firmware 2.1.10, 2.1.11, and 2.2.0 each uses a different Ethernet library. So if none of them works, I have no idea what it could be. Maybe it's the Ethernet module itself. You can try to swap the Ethernet modules between the controllers and see what happens.

sprinkmeier, 2023-01-16:
Are the devices getting IP addresses from a DHCP server?
Are the logs from that server available?
Can you keep the controllers on the network by pinging them?

KasKam, 2023-01-16:
We have been swapping out the ethernet devices and using a script to ping them. So far it seem that changing out the ethernet devices has helped a lot. If they stay as stable as they have been we will try removing the ping as well.

KasKam, 2023-01-24:
We believe this was fixed by switching out the Ethernet modules on the units. We had several units burn out from running to many solenoids at once and we believe this damaged the Ethernet modules as well. We never changed them after switching out the burned out units.  Thanks for your advice.

closed 2023-01-24 by KasKam