OpenSprinkler/OpenSprinkler-Firmware #231: mqtt api

foscoj, 2023-05-16:
Hi,

the mqtt only shows online/offline and after a reboot it does not switch to online immediately.

I see the mqtt-api branch, but it seems to be abandoned.

Are there any plans to implement a full mqtt api support?

At least for status and control of starting and stopping valves/programs would be great for UI and remote control button support.

Thanks!

rayshobby, 2023-06-05:
The entire mqtt-api branch has been integrated into the latest firmware. We tested MQTT and everything is working as expected. Did you subscribe to a specific topic or have you tried to subscribe to wildcard?

foscoj, 2023-06-05:
I tried to inspect the data with the MQTT Explorer, MQTT Snooper on android results in the same view.
I am on a wired connection with the 3.2 AC Hardware.
MQTT integration is enabled to a mosquitto on raspberry without authentication.
[image]
[image]

foscoj, 2023-06-05:
ah, when I activate a station, a mqtt message is available. But it stays online with JSON content.
It seems to be, that the REST API principles are used for mqtt communication, this is a little bit strange ;)

foscoj, 2023-06-05:
ok, so it is json notifications in mqtt currently.

thanks, but not really what I expected :(

rayshobby, 2023-06-05:
If you want to change it to other formats, you can modify the firmware yourself. The MQTT feature was implemented by other developers and they had a reason to implement it in JSON format.

closed 2023-06-05 by rayshobby