OpenSprinkler/OpenSprinkler-Firmware #232: Feature Request - Allow Zone Reordering

kars85, 2023-05-28:
I know this can be accomplished through hardware (rewiring zones), but I think it would a nice feature to be able to reorder zones through the UI. For example, I'd prefer all of my front yard be watered in the early morning hours before heavy foot traffic, then my backyard can run after that for as long as necessary. However, the installer from the previous owner set it up for the zones to run in a big circle round my property.

Is that something that you could incorporate into a future update?

Thanks!
Karson

rayshobby, 2023-06-05:
This would require a major change to the firmware and also the UI. There is a work around you can use, but it does require having some spare zones. The work around is to use multiple zone ports for the same physical valve. For example, let's say you have 4 valves, v1, v2, v3, v4, you can connect v1's zone wire to both zone ports 1 and 8; similarly v2's zone wire to both ports 2 and 7 and so on. It is such that zones 1 and 8 both activate the same physical valve v1; zones 2 and 7 both activate the same physical valve v2 and so on. Let's say in the morning you want to water in the order of v1->v2->v3->v4, in the afternoon you want to reverse it v4->v3->v2->v1, you can set two programs, the morning one will turn on zones 1, 2, 3, 4, and the afternoon one 5, 6, 7, 8. This will allow you to achieve the reordering, but it does waste some zone ports, and if you have a large number of valves you will need zone expander.

rayshobby, 2024-07-29:
As an update, there is an initial support for this feature in branch dev/221_1:
https://github.com/OpenSprinkler/OpenSprinkler-Firmware/tree/dev/221_1
it's not a support for arbitrary re-ordering, rather, it supports a few pre-defined zone ordering methods in a program, such as reverse the order, alternate the order (all odd-numbered runs forward, all even-numbered runs backward) etc. It can also order zone runs in a program sorted by zone names, or use a random ordering.

kars85, 2024-07-29:
Thanks for the update!Sent from my iPhone

https://github.com/OpenSprinkler/OpenSprinkler-Firmware/tree/dev/221_1

wernerdaehn, 2024-09-08:
I believe this is a must have feature, in the global overview and per program.
In my case I have 6 boxes with 2-3 valves. One valve controls the set of sprinklers and the others water some plants. Some are not used yet.
Obviously I want to see in the overview the sprinklers first, the plants second and the not-used once at the end or not at all. Each box is connected by GND and CH1,CH2,CH3. So S1 is CH1 for the first box, S4 is CH1 for the second box etc.

zedzedzed, 2025-03-20:
> As an update, there is an initial support for this feature in branch dev/221_1: https://github.com/OpenSprinkler/OpenSprinkler-Firmware/tree/dev/221_1 it's not a support for arbitrary re-ordering, [...]

Hey @rayshobby, thanks for the firmware update. I've applied it but I think I'm blind as I don't see zone ordering options when I edit a program. Can you please point me to where I can order the zones?

```
App Version: 2.4.49
Firmware: 2.2.1 (1)
Hardware Version: 3.3 - AC
```

rayshobby, 2025-03-20:
The feature is implemented by annotating the program name. Please check the user manual for details.

robohack, 2025-04-23:
I can't find any mention of this feature in the v2.2.1 user manual.  I had to read the source to figure out how to do it!

The only ":>" program name annotations I can find (by searching for ":>") mentioned in the manual are in the "Special program names" section on page 14 and are for ":>reboot" and ":>reboot_now".

robohack, 2025-04-23:
That is indeed a different manual than the one I downloaded yesterday.

Not sure how I could have got a different copy yesterday though when the web page says it was updated on "6 Mar, 2025 at 10:08 AM"!  Unless maybe my finger slipped as I was clicking on the link.

Anyway, I see the right manual now, sorry for the confusion!