OpenSprinkler/OpenSprinkler-Firmware #235: Feature Request: Add new Zone Attribute "Individual" + Max number of parallel zones

Vrumfondel, 2023-06-05:
Hello!
In my sprinkler setup I would need a feature that one of the Zones is allowed to only run alone, while any for the others any two can run together.

Currently I don't see how I could realize this config - so the only choice is to run any of them individually (which is neither time nor power efficient).

Request:
1. Add Zone Attribute to allow Zones to only run by their own
2. Add configuration for max Zones to be run parallal

rayshobby, 2023-06-05:
Firmware 2.2.0 has a feature called sequential groups. Have you looked into that? You can always create multiple programs to achieve what you want.

Vrumfondel, 2023-06-05:
I know the sequentiell groups, but as I see it is not possible to realize what I want.Multiple programs would be a workaround, but not a real solution.

rayshobby, 2023-06-05:
I honestly don't understand what you would like to implement. What do you mean by "allow Zones to only run by their own"? Isn't this what 'sequential' means? That is, zones will be serialized and run one after another. perhaps you can give a specific example of how you'd like zones to run.

In the end, since you have multiple programs, you can always make programs to run in the way you want your zones to run. I guess what you wanted is more of a way for the firmware to automatically schedule zones according to some specific restrictions. But again, I don't quite understand exactly what you need. Therefore an example would be more useful.

Vrumfondel, 2023-06-06:
OK, let't do an example:
There are 4 sprinkler circuits, 1,2,3,4 .
 
Circuit one must only run alone due to waterflow availability. Any two of circuits 2,3,4 can run together, but not all 3.

Sure I can do this with multiple programs, but having a smart solution like opensprinkler I prefer to only define on start- and the individual sprinkler-times, and the rest shall be handled by the controller.

In addition to this, there might be electrical limitations on the controller side, so I'm missing for a parameter to limit the number of valves to be driven at one time - independent of any other "parallel/sequential" configurations.

rayshobby, 2023-06-06:
So in your example, can 1 run in parallel with 2, 3, 4? For example, can 1 run at the same time with 2 and 3, so all of 1, 2, 3 are running at the same time?

Vrumfondel, 2023-06-06:
No. For 1 no other parallel circuit ist allowed.Possible combinations are:12+32+43+4One other possibility to implement this would be giving each circuit a percentage in how much it consumes of the total possible waterflow of the source. The planning algorithm ensures 100% are never exceeded.

rayshobby, 2023-06-06:
OK, I am very confused. I don't know how this can be translated into program code to be honest. Maybe you can find a programmer to help you.

closed 2023-06-06 by rayshobby