OpenSprinkler/OpenSprinkler-Firmware #242: Feature: Resilience building - Loading UI for OSPI-Webserver without ui.opensprinkler.com

amnesiemaniac, 2023-07-23:
Hi
After my first build I started OpenSprinkler.service and loaded http://[host]:8080. I got an empty site.
What happend: I'm using an addon "No script" in my brwoser to control what I site does on my computer and this blocked loading from ui.opensprinkler.com. Without that OSPI is not working.

So If ui.opensprinkler.com is unavailable, I can't use OpenSprinkler.
This was an "Van Moof" - Moment for me - you know that expansive bikes? Van Moof is an insolvent company.
If they shut down company servers, People can't use bikes bought anymore. I think this is not a good situation, a too deep dependency.

Feature Request:
Bring UI-Funtions in Firmware or in a deb-Package or anything like this. A OpenSprinkler should be a usefull device with and without internet access, without dependency to anybody.

Thanks!

salbahra, 2023-07-23:
Hi Michael,

I'm the author of the application and just wanted to comment that the UI can be loaded from any source such as the mobile applications (iOS or Android) which do not require any internet connection. They only need local network access to communicate. Furthermore, the UI can be saved to any local web server and used to access the device that way as well.

Bundling the UI inside the firmware was classically an issue due to size restrictions on these smaller devices. It's also more difficult to keep things up to date as the UI and firmware update at different frequencies.

The UI can be downloaded from here: https://github.com/OpenSprinkler/OpenSprinkler-App/tree/master/www

Hope this helps!

Samer

amnesiemaniac, 2023-07-25:
Thanks @salbahra !

I just looked on my raspi 4 and did't see how many different small devices you support.

I'll try to change my OSPI to use a local copy of the UI.
Maybe we can make a little HowTo to support people who wants local UI? Is anybody interested?

I've thought about what would be necessary:
1. webserver (nginx on my OPSI / Raspi or any other server), config to load UI
2. config for OSPI firmware to stop build-in webserver on Port 8080
3. Description to handle Firmware upgrades

What else?

rucksman, 2024-06-07:
@amnesiemaniac Did you succeed in hosting all the necessary files on a private webserver? If yes, how did you do it? I read somewhere there is a hidden URL http://<ip-of-opensprinkler>:8080/su where you can change https://ui.opensprinkler.com/js to a different path. But this does not work as only home.js can be hosted on my own webserver, app.js and app,css etc. will be loaded from some weird paths which fails and makes no sense for me to mimick these paths. Not only for resilience, but also for privacy reasons it would be great to host the UI on a private webserver.

rayshobby, 2024-07-25:
How to customize the UI / hosting UI on a different server is explained in this support article:
https://openthings.freshdesk.com/support/solutions/articles/5000164006

closed 2024-07-25 by rayshobby

rucksman, 2024-08-15:
Sorry to get to this topic again, but the solution is not working, at least not for me. I have the "OpenSprinkler Pi" since some years working on the IP address 192.168.50.60. The UI is therefore available on 192.168.50.60:8080.

I now basically followed your Option 1 from [here](https://openthings.freshdesk.com/support/solutions/articles/5000164006-customize-the-ui-app). I have an internal webserver running at home, so I did put the contents of the www folder into a new folder on my webserver. Now when I access this path, I get this:
[image]
When trying to scan, it finds nothing (both the Raspberry Pi and the webserver are on the same network):
[image]
When I try the manual way (entering 192.168.50.60 into "OpenSprinkler IP", it also fails:
[image]
What am I doing wrong?