OpenSprinkler/OpenSprinkler-Firmware #243: Manually starting zones from API does not follow sequential rules

kamakazie2, 2023-07-24:
Hi all,

I have these 2 zones setup in Sequence group C and when i call via the API they will both turn on where i would expect the behavior to be along the lines of 4000s on SID 11 with SID 12 waiting until SID 11 is complete.. 

		sendHttpGetRequest("http://192.168.0.55/cm?pw=xxx&sid=11&en=1&t=4000")
		sendHttpGetRequest("http://192.168.0.55/cm?pw=xxx&sid=12&en=1&t=1000")

When i do this in the GUI it works as expected. where SID 12 is yellow and waiting for SID 11 to complete.

Is this the intended usage of the API call to ignore sequential groupings?
Thanks!

rayshobby, 2023-07-24:
I honestly don't know. The GUI uses exactly the same /cm API to turn on zones manually, so there shouldn't be any difference. What scripting language are you using? Where is that `sendHttpGetRequest` function from?

rayshobby, 2023-07-24:
I just tested this on `demo.opensprinkler.com` and I cannot reproduce the issue. For example, I can use:
`curl "https://demo.opensprinkler.com/cm?sid=1&en=1&t=4000"`
then
`curl "https://demo.opensprinkler.com/cm?sid=2&en=1&t=1000"`
and the second zone is queued waiting to run after the first zone. I do not see them running together.

kamakazie2, 2023-07-24:
I just did the same manually and it worked. 
Im using a script from my openhab home automation to trigger open sprinkler events.
I probably need to put a delay between commands. Let me put a delay in between commands and ill report back.

kamakazie2, 2023-07-24:
That was it. I put in a 1s delay and it worked.

closed 2023-07-24 by kamakazie2

rayshobby, 2023-07-24:
OK, that's interesting, I wouldn't except the delay to make a difference, because the firmware processes the API commands in sequence (it's a single thread processor). But in any case, at least it works now.