OpenSprinkler/OpenSprinkler-Firmware #244: Channel 1 only runs for a few seconds.

icemansid, 2023-07-25:
I have an issue that I cannot seem to resolve.  Channel 1 only runs for a few seconds regardless of what the run-time is set.  I have reloaded the software and the issue persists across multiple versions.

[image]
[image]

icemansid, 2023-07-25:
OpenSprinkler Pi, Raspberry Pi0, Firmware 2.2.0, App Version 2.3.3.

HomeAssistant has been configured in the past but the integration is currently disabled.  I also have another controller that I was using as a master controller but I have been trying to eliminate as many things as possible before asking for help.

Is there a log that I could look through to see if it is getting a shut-off command from some other location?

rayshobby, 2023-07-25:
If you suspect something else is sending command to turn it off, you can change the device password to prevent other scripts from sending command to it. Also make sure the Ignore Password option is not checked.

rayshobby, 2023-07-25:
You said firmware 2.2.0, which minor revision? You also said you've tried a few different firmware versions and they all have the same issue?

icemansid, 2023-07-25:
2.2.0 (2).  I do not know what the previous version was but it was setup over a year ago.

The install process I ran was form this page.

https://openthings.freshdesk.com/support/solutions/articles/5000631599-installing-and-updating-the-unified-firmware-on-ospi

rayshobby, 2023-07-25:
So even if you manually run zone 1 for 1 minute, it only runs 4 seconds? The firmware running at
demo.opensprinkler.com
is firmware 2.2.0(2). I cannot reproduce the issue. You can try it yourself there.

icemansid, 2023-07-25:
Is there a way I can view more detailed logs that may show why its shutting off?  I am unable to post anything on the forums for this app and Support is refusing to talk through a ticket.

rayshobby, 2023-07-25:
The reason we cannot provide support to you is that you didn't buy any product from us. We only provides support to customers who bought our products.

icemansid, 2023-07-25:
sorry - running for 1 minute ran for 1 minute. 5 minutes ran for 2m17s this time.  This morning, was 27 seconds for a 15 minute schedule.

As I tried to say in the ticket I would use the Forum if it worked. every link ends in a page not found.

icemansid, 2023-07-26:
Closing this issue, I am working around the problem by reassigning the GPIO pin to station 3.  Even when station 1 is reset and not assigned to anything, it still shuts off randomly.

closed 2023-07-26 by icemansid