OpenSprinkler/OpenSprinkler-Firmware #246: OpenSprinkler does not compile with -std=c++17

lastewart, 2023-08-03:
Compiling on raspbian 'bookworm' fails:

```
$ g++ --version | head -1
g++ (Raspbian 12.2.0-14+rpi1) 12.2.0

$ g++ -o OpenSprinkler -DOSPI main.cpp OpenSprinkler.cpp program.cpp opensprinkler_server.cpp utils.cpp weather.cpp gpio.cpp etherport.cpp mqtt.cpp -lpthread -lmosquitto
In file included from OpenSprinkler.h:29,
                 from main.cpp:26:
utils.h:46:1: error: reference to ‘byte’ is ambiguous
   46 | byte file_read_byte (const char *fname, ulong pos);
      | ^~~~
In file included from /usr/include/c++/12/string:42,
                 from defines.h:494,
                 from OpenSprinkler.h:28:
/usr/include/c++/12/bits/cpp_type_traits.h:406:30: note: candidates are: ‘enum class std::byte’
  406 |   enum class byte : unsigned char;
      |                              ^~~~
defines.h:29:23: note:                 ‘typedef unsigned char byte’
   29 | typedef unsigned char byte;
      |                       ^~~~
[...]
```

Seems that the g++ version on bookworm defaults to `-std=c++17` which introduces the duplicate typedef for `byte`. Compiling with `-std=c++14` works around the problem:
```
$ g++ -std=c++14 -o OpenSprinkler -DOSPI main.cpp OpenSprinkler.cpp program.cpp opensprinkler_server.cpp utils.cpp weather.cpp gpio.cpp etherport.cpp mqtt.cpp -lpthread -lmosquitto
```

Alternatively, you'll need to conditionalise the code to use the `c++17` definition of `byte` when it's available.

n3xus1, 2023-10-18:
Looks like this error is still present, it would be nice to push this temp fix into build.sh until a permanent solution is found and implemented.
Thanks for the temp solution.

rayshobby, 2023-10-18:
The fix is already in the support/os33 branch and will be merged to master soon;
https://github.com/OpenSprinkler/OpenSprinkler-Firmware/tree/support/os33

closed 2023-10-18 by rayshobby