OpenSprinkler/OpenSprinkler-Firmware #249: Ethernet connection lost

hanjo, 2023-09-01:
Hi,

my device:
Firmware: 2.2.0 (1)
Hardware Version: 3.2 - AC

Since some time, every few days Opensprinkler disconnects from the network. In fact, other than the network the device seems to be working fine, as the schedule still works. When clicking the first button on the device, it shows an IP address of 169.254.something, which tells me that it somehow wasn't able to refresh the IP address with the DHCP server. However, no other device on my network has this problem.
The only way to enable the ethernet again that I found so far is to power cycle the device. Most of the time, Opensprinkler will get an IP from the DHCP after a power cycle, but not always. If it doesn't, it falls back to WiFi mode and I have to power cycle again to fix it. If this happens, it behaves like there is not Ethernet module connected at all.
After a few days then, the cycle starts all over...

I cannot exactly pinpoint when this behavior started, but I'm pretty sure I did not have it before I updated to 2.2.0, so it could be related to the new network stack. I should probably mention that my ethernet module seems to be ENC28J60, as far as I can tell.

This also seems similar to https://github.com/OpenSprinkler/OpenSprinkler-Firmware/issues/218

I appreciate any suggestions.

Thanks
Greetings
Hanjo

jabums, 2023-09-03:
Since I purchased  my Opensprinkler 3.2 - DC in July,  I have had exactly the same problems as described above.
Unfortunately I can not be of any help concerning your problems hanjo.

uneart, 2023-10-26:
@hanjo there seems to be an inofficial firmware version 2.3.0 (132) by the german distributor of OpenSprinkler @opensprinklershop which includes some adjustments to the ethernet / DHCP handling (see [changes](https://github.com/OpenSprinkler/OpenSprinkler-Firmware/compare/master...opensprinklershop:OpenSprinkler-Firmware:master)). Maybe it helps. I successfully updated so far without any issues.

**IMPORTANT: You loose official vendor support using this firmware, so use at your own risk. Not sure, how well downgrades work.**

pshempel, 2023-11-16:
I have had the same issue described above using: 
Firmware: 2.2.0 (2)
Hardware Version: 3.2 - (AC)

The device is working and runs program schedules. Will lose ethernet connectivity after some time.

hanjo, 2023-11-16:
I also installed the firmware version 2.3.0 from https://opensprinklershop.de/2023/10/04/firmware-2023-2-3-0/ a few weeks ago, and so far I had no more disconnects. This seems to have fixed the issue. I will leave this issue open though, because it is not fixed in the original firmware.

phoenix-9, 2024-04-10:
I have experienced loss of connectivity also.  I use wired ethernet. Version 2.2.0 (3).  Manually configuring static IP is my work-around.

tcpdump will show that opensprinkler is requesting a DHCP lease.  The DHCP server responds with an address.  Opensprinkler never acknowledges and will request another lease a few seconds later.

I have not figures out why it happens- seems random.  A power cycle is required to resume normal operations.

opensprinklershop, 2024-04-11:
OpenSprinkler is based on Arduino. Latest Arduino firmware update fixt many
issues. You can try the new 2.3.1(151) from my website. Another option is
to use a W5500 Module and the OS 3.2 Adapter for better connectivity.

Best regards
Stefan Schmaltz
OpenSprinklerShop (Germany)

hanjo, 2024-04-11:
@opensprinklershop
> Another option is to use a W5500 Module and the OS 3.2 Adapter for better connectivity.

This sounds like a great alternative - may I suggest to offer this for purchase in the shop then?

starsoccer, 2024-05-08:
+1 to the above. It seems like this issue is more or less the same as this thread too, https://opensprinkler.com/forums/topic/controller-lockups-crashes

Just to confirm what the best path forward is for users experiencing it, like myself, it seems like there are 2 options:
1. Upgrade to the latest unofficial firmware from here, https://opensprinklershop.de/en/neuigkeiten/
2. Buy a W5500 Module and then some OS 3.2 Adapter. It seems like one is sold here, https://opensprinklershop.de/en/product/nachruestsatz-ethernet-fuer-opensprinkler-3-3-w5500/ but is only for Open Sprinkler 3.3. Its also not clear if/what adapter one would need?

opensprinklershop, 2024-05-08:
regarding 2) this link is a complete W5500 + enclosure + cable for OS 3.3. You also need a 3.3->3.2 cable adapter to connect the W5500. Just add a notice with your order, I can add the cable adapter without extra charge.

starsoccer, 2024-05-08:
> regarding 2) this link is a complete W5500 + enclosure + cable for OS 3.3. You also need a 3.3->3.2 cable adapter to connect the W5500. Just add a notice with your order, I can add the cable adapter without extra charge.

Okay noted thanks. Sadly it doesnt seem like you ship to the US though. Any chance I can message you to get one?

rayshobby, 2024-05-08:
Here are some clarifications:
1. OS 3.2 uses ENC28J60, OS 3.3 uses W5500. These two modules have different pin ordering so they are not directly interchangeable. The firmware supports both types of Ethernet modules, but to avoid confusion when people purchase Ethernet modules separately, we made it such that if you have OS 3.2, then use ENC28J60, if you have OS 3.3, then get W5500.
2. Because the firmware supports both, technically you can use W5500 on OS 3.2 too, but it would require a small pin-change adapter, that re-routes the pins of W5500 to match that of ENC28J60. We don't officially sell this adapter because it's custom made, and making it available can introduce further confusions (believe it or not, people don't even read the existing user manual, so getting them to read anything else is even harder).
3. If you want W5500 but you have OS 3.2, you can send a support ticket to support.opensprinkler.com and arrange for either of two options: I would recommend simply replacing the top board to OS 3.3 so you can directly use W5500 (the bottom driver board does not need to be replaced) and there is less confusion; another option is to get a W5500->ENC28J60 pin-change adapter, as explained above.

starsoccer, 2024-05-08:
> Here are some clarifications:
>
>     1. OS 3.2 uses ENC28J60, OS 3.3 uses W5500. These two modules have different pin ordering so they are not directly interchangeable. The firmware supports both [...]

Thanks for clarifying. I just opened a ticket. Ive only been using my unit for a few weeks since I turned on my sprinklers but have had a few lock outs already.

hanjo, 2024-05-12:
Just for anyone following this: I have received a W5500 from @opensprinklershop along with the PIN adapter that makes the chip compatible with OS 3.2. Since then, I had no more disconnects and everything seems to be working fine.

While this does not solve the underlying issue with the ENC28J60, this fixes the immediate problem of network connectivity issues. Since I'm not sure if the issue with the ENC28J60 can even be fixed, but a workaround exists now, I'm closing this issue.
Thanks for all the help!

closed 2024-05-12 by hanjo