OpenSprinkler/OpenSprinkler-Firmware #250: Recent kernels no longer support sysfs interface for GPIO (raspberry pi)

jbalonso, 2023-09-01:
See https://forums.raspberrypi.com/viewtopic.php?t=343514.

On my recent system, running `Linux sprinkler 6.3.9-1-default #1 SMP PREEMPT_DYNAMIC Thu Jun 22 03:53:43 UTC 2023 (0df701d) aarch64 aarch64 aarch64 GNU/Linux`, the sysfs interface no longer works. This breaks `raspi-gpio`, and I _think_ this critically impacts the workings of https://github.com/OpenSprinkler/OpenSprinkler-Firmware/blob/master/gpio.cpp, beyond even just the `system()` call to `raspi-gpio`.

A potential remedy is written up here: https://www.ics.com/blog/gpio-programming-exploring-libgpiod-library. There exists packages for `gpiod` (CLI tools), `libgpiod2` (runtime libraries), and `libgpiod-dev` (runtime libraries and development headers).  I notice three choices:
1. Replace `raspi-gpio` and the sysfs interface with the use of `libgpiod`.
2. Replace `rapsi-gpio` and the sysfs interface with `system()` calls to `gpioset`.
3. Continue supporting older kernels by conditionalizing the use of `raspi-gpio` and the sysfs interface vs. `gpioset`.

I might be able to do a little empirical testing to see if `raspi-gpio` and `gpioset` both work on older kernels.

jbalonso, 2023-09-05:
I did not intend to close this issue when working on it in my fork... sorry.

jbalonso, 2023-09-05:
I worked on this over in my fork (please see the PR in my fork that accidentally closed this issue). I'd have offered it as a PR here, but:
1. My fork builds off of @developer79433's #225 PR, so the dependencies are corrected in `Dockerfile` but not `build.sh`.
2. Rather than porting the preliminary work on interrupts to libgpiod (which is possible), I just deleted it. Even then, I did not remove the pthreads includes/linking, though I _think_ I determined they were only present for the interrupt handlers.
3. I didn't address the possibility of needing to run this firmware on an older kernel that supports the sysfs interface (as well as `raspi-gpio` 's peculiar manipulations of `/dev/gpiomem` or `/dev/mem`).

I did, however, determine the code went from not-working to working by actually watering my lawn.

lnxsrt, 2024-01-20:
Your code tweaks worked for me. However libgpiod in ArchLinuxARM is now at version 2.0.1 which completely breaks the older API. Downgrading to libgpiod 1.6.4 allowed your code to compile. Just lovely how GPIO handling in the kernel and libraries seem to change every few months. Figured by now those APIs would be fairly stable, but change for change sake I guess. Thanks again!

jbalonso, 2024-01-22:
Thanks for the heads-up!

Looking at distros, it seems Ubuntu (the base image I used for the docker image), Alpine (what I probably should have used), and even openSUSE Tumbleweed (what my k8s cluster running opensprinkler for me uses natively) are all still at 1.6.x.

I confirm, though, that Arch Linux is at 2.0.1.

It looks like I should bring my fork up-to-date with this repo anyway, and I should see if I can support both old and new `libgpiod` with good-old preprocessor directives. Unfortunately, it's the dead of winter here, so this will likely wait until spring is closer. (Also: a quick investigation shows that ArchLinuxARM is going to be tricky to test against).

Separately: you could try running opensprinkler as a container using docker or containerd (`nerdctl`) and side-step downgrading your system `libgpiod`.

wommy, 2024-04-18:
i got it working on bookworm by running the following commands

``` bash
wget https://raw.githubusercontent.com/jbalonso/opensprinkler-firmware/main/gpio.cpp \
-O gpiod.cpp # pulls @jbalonso's edited gpio.cpp renaming it gpiod.cpp

cp build.sh build_bw.sh # perserve executable
sed -i '26s/-dev/-dev libgpiod-dev/' build_bw.sh # avoid a second query by appending
sed -i '/raspi-gpio/,+5d' build_bw.sh # delete raspi-gpio and its command check
sed -i '/-DOSPI/s/gpio/gpiod/' build_bw.sh # in the `-DOSPI` line, change gpio to gpiod
sed -i '/-DOSPI/s/$/ -lgpiod/' build_bw.sh # append ` -lgpiod` 

sudo ./build_bw.sh ospi # and finally run the build script
```

mrave, 2024-04-26:
@wommy followed your instructions, still not working, any tips, or things to check?

jbalonso, 2024-04-26:
I suspect we'll all be happier if we added a github action to this repo to _build_ the binary as well as the docker image. I wonder if we can assure that the `libgpiod2` library can be statically linked...

rayshobby, 2024-04-26:
Sorry about the delay. I will take a look at this soon. We are still in the middle of the semester and I don't have time to check all the details. The semester will be over in a few weeks by which time I will have more time. To be honest I don't understand why every couple of years Raspberry Pi decides to change the way gpios are operated. Instead of providing a backward compatible method, they just get rid of the old method. It's like not understanding what software abstraction means -- it should provide a high level interface that remain unchanged over decades.

mrave, 2024-04-26:
> I suspect we'll all be happier if we added a github action to my repo to _build_ the binary as well as the docker image. I wonder if we can assure that the `libgpiod2` library can be statically linked...

That would be great indeed... but I still wonder whats not working, since I can't see any logs or anything. Any idea?

> Sorry about the delay. I will take a look at this soon. We are still in the middle of the semester and I don't have time to check all the details. The semester will be over in a few weeks by which [...]

@rayshobby thanks for the prompt reply. The real problem here is that the irrigation is not working for a week (since I upgraded raspibian) and i did not even know, until I noticed the plants getting sick... 
Anything I can do to help? Is there any was I can run the taps from command line, i.e. until this gets fixed? If there's an easy tool to talk to the electric taps I can just cron job it for now.

wommy, 2024-04-26:
> Does it make sense to clone the `jbalonso` repo entirely and run it instead, is that recommended?

not at all, unless youre fluent in docker. 

> @wommy followed your instructions, still not working, any tips, or things to check? 

did you run those commands on the repo after a pull?

wommy, 2024-04-26:
tomorrow, im going to reinstall raspian bookworm to test this, as i might be missing a dependency 

https://github.com/wommy/OpenSprinkler-Firmware/commit/981b2151b6f2665b07a9209aec592c85af4b996a

mrave, 2024-04-27:
> @mrave
> 
> > Does it make sense to clone the `jbalonso` repo entirely and run it instead, is that recommended?
> 
> not at all, unless youre fluent in docker.
> 
> > @wommy followed your instructions, still not working, any tips, or things to check?
> 
> did you run those commands on the repo after a pull?
> 
> your else block should look like this
> 
> ```shell
> else
> 	echo "Installing required libraries..."
> 	apt-get update
> 	apt-get install -y libmosquitto-dev libgpiod-dev
> 	echo "Compiling firmware..."
> 	g++ -o OpenSprinkler -DOSPI -std=c++14 main.cpp OpenSprinkler.cpp program.cpp opensprinkler_server.cpp utils.cpp weather.cpp gpiod.cpp etherport.cpp mqtt.cpp -lpthread -lmosquitto -lgpiod
> fi
> ```

Thanks @wommy. I ended up cloing your repo, fixing the missing libgpiod, std arguments, rebuilding and using the `OpenSprinkler` binary for now. So I am ok for now, until @rayshobby will incorporate this into the master. Appreciate your help man!

rayshobby, 2024-04-28:
> That would be great indeed... but I still wonder whats not working, since I can't see any logs or anything. Any idea?
>
>
> @rayshobby thanks for the prompt reply. The real problem here is that the [...]

Why not downgrade Raspbian to Bullseye? Not sure if you noticed, but we've put a note on the instructions page:
https://openthings.freshdesk.com/support/solutions/articles/5000631599-installing-and-updating-the-unified-firmware-on-ospi
that says specifically the firmware currently does not work with Bookworm. Unfortunately OSPi does not have current sensing ability so it doesn't show you the current draw from solenoids. If you have OpenSprinkler 3.x, they will show the actual current draw, which would allow you to see that the solenoids are not open. On the other hand, you can install a flow/water meter and use it to monitor the flow rate. This would allow you to notice that no water is flowing.

mrave, 2024-04-28:
Yeah, downgrading debian is, hmmm, not recommended anyway :) 
I am running currently with @wommy code and it seems to work. I'll wait till you have some time to upgrade formally.

rayshobby, 2024-05-11:
I will have some time the next couple of weeks to take a look at this. I have a couple of quick questions:
1. Does gpiod work with older Raspbians? I guess I could try it out myself, but someone may know it already, and it would be good to have one unified method for all Raspbians.
2. How does gpiod handle interrupts? The flow sensor requires interrupt and I took a quick look at:
https://raw.githubusercontent.com/jbalonso/opensprinkler-firmware/main/gpio.cpp
and saw that the attachInterrupt function is currently empty.

Honestly I don't know why it's so difficult for RPi foundation to just keep one consistent method for GPIO controls that they don't get rid of every few years. Most people I know use RPi purely for embedded projects, and for embedded projects GPIO controls are the most basic functionality. Arduino has been around for decades and their basic GPIO controls (pinMode, digitalWrite, digitalRead, attachInterrupt) have remained the same since the beginning. This makes it so much easier to migrate software. I am sure this whole getting rid of sysfs thing has confused lots of developers and angered some of them too.

rayshobby, 2024-05-12:
Never mind, I was able to incorporate the changes. I grabbed the changes made by:
https://github.com/opensprinklershop/OpenSprinkler-Firmware
which was itself based on jbalonso's work. Thanks to both.

The updated code is currently at this branch:
https://github.com/OpenSprinkler/OpenSprinkler-Firmware/tree/support_rpi_gpiod
I will leave there for a few days so users (and us internally) can test before deploying.

I apparently forgot the flow sensor no longer uses interrupt (I guess I must have removed it due to multiple issues using interrupt). So the fact that there is no support for interrupt currently is ok, since it's not being used. This also implies that technically we can just use raspi-gpio command since it supports basic digital pin functions and works on all Raspbian versions (though I suspect using it for flow sensor reading might be too slow).

jbalonso, 2024-05-13:
> 1. Does gpiod work with older Raspbians? I guess I could try it out myself, but someone may know it already, and it would be good to have one unified method for all Raspbians.

Probably not, but if you frame the question in terms of _supported_ Raspbians (those with recent/supported Linux kernels), probably yes.

> 2. How does gpiod handle interrupts? The flow sensor requires interrupt and I took a quick look at:
>    https://raw.githubusercontent.com/jbalonso/opensprinkler-firmware/main/gpio.cpp
>    and saw that the attachInterrupt function is currently empty.

I could tell that the interrupt code wasn't being used, but I thought it was preliminary rather than removed (which you noted in your subsequent comment). In any case, I just removed it rather than porting it.

> Honestly I don't know why it's so difficult for RPi foundation to just keep one consistent method for GPIO controls that they don't get rid of every few years. Most people I know use RPi purely for [...]

Truthfully, the gpiod-vs-sysfs thing happened in the mainstream Linux kernel and reaches far beyond just the RPi. (Note: I didn't do research to see if the RPi Foundation _instigated_ it). I'd argue that the RPi CLI tools for gpio fiddling should have been maintained and ported, but even I have a hard time arguing for that when the libgpiod user mode library also includes much more sophisticated CLI tools (if I recall correctly).

rayshobby, 2024-05-13:
> Probably not, but if you frame the question in terms of _supported_ Raspbians (those with recent/supported Linux kernels), probably yes.

I adopted Stefan's changes, which keeps both sysfs and gpiod in the code, and uses precompilation macro to switch between them depending on the existence of "/sys/class/gpio/gpiochip512/". This detection may be a bit ad-hoc, since I am not entirely sure whether this is a good way to check whether sysfs gpio control is available or not, but I can investigate further.

> I could tell that the interrupt code wasn't being used, but I thought it was preliminary rather than removed (which you noted in your subsequent comment). In any case, I just removed it rather than [...]

So it sounds like it's not obvious / trivial to use interrupts with gpiod. I wish there was a simple way like attachInterrupt in Arduino. 
 
> Truthfully, the gpiod-vs-sysfs thing happened in the mainstream Linux kernel and reaches far beyond just the RPi. (Note: I didn't do research to see if the RPi Foundation _instigated_ it). I'd argue [...]

Yes I am aware this change happened in mainstream Linux kernel, and I don't blame RPi for making the changes. But what puzzles me is why can't they provide a C library, perhaps built on top of sysfs or gpiod or whatever low-level library they choose to build on, which has consistent API that doesn't change over time. I thought wiringPi was a great idea and it mimics Arduino (I know, they didn't write it, but they could have taken over it). It's akin to say, now Mac has moved from PowerPC chips to Intel chips, your code, including how summation and multiplication work, all must change. And we will do this again in a few years time when we transition from Intel chips to Apple silicon. This sort of breaking changes is what makes me want to stop developing for RPi and just abandon the whole OSPi project.

What I needed (and I think a lot of other embedded developers) is really just pinMode, digitalWrite, digitalRead and attachInterrupt, which doesn't come and go when I upgrade Raspbian. (I2C and SPI too if possible but those can be bit-banged in the worst cast). Is this too much to ask? I don't think so.

closed 2024-05-13 by rayshobby