OpenSprinkler/OpenSprinkler-Firmware #253: Programs with interval are not executed correctly by the run Feature

AlexanderNikopol, 2023-10-01:
If a program has set "repeat every" and "repeat count" and this program is executed by "run <program_name>", the interval is ignored.
The manual says, if using this run once feature, the programs will be executed as configured.

Taking a look at the code:
The web GUI sends an array to the opensprinkler device. This array keeps only the duration time for watering of each station for the corresponding program. There is no information about repeats and interval.

AlexanderNikopol, 2023-10-01:
It was tested with the 220(2) tag.

rayshobby, 2023-10-16:
The Run-Once feature uses OpenSprinkler's run-once API which does not include any repeated program runs. At the moment, the work-around is to duplicate the program and set a start time that's the current time plus 1 minute, so it can start within 1 minute and since it's a set program it will be able to do repeats.

The bottomline is that if repeated runs are needed, it needs to be a program that's saved in flash memory. The run-once (as the name suggests, it only runs once) does not save the program therefore it does not handle repeats.

In the future we can add a feature that supports repeated runs. That would be implemented by essentially saving it temporarily as a program, and removing it after all the repeats are finished.

closed 2023-10-16 by rayshobby