OpenSprinkler/OpenSprinkler-Firmware #255: Weather service data error

sumnerboy12, 2023-10-08:
Hi,

I am running the latest firmware on my OpenSprinkler;

[image]

After rebooting the device connects to the weather service fine, and updates my watering level etc. However after a day or two, without fail, I get the error below (no other details);

[image]

From this point on the watering level doesn't update and I have to reboot the device to restore the connection to the weather service.

I have tried the suggestions in [this](https://opensprinkler.com/forums/topic/weather-server-call-failures/) forum post but switching to Stefan's weather service made no difference, with the same errors occurring.

This is becoming a problem as my watering level is getting stuck on whatever it was the day the error starts happening, and if I forget to check and reboot I can end up over/under watering quite a lot!

There are quite a few details in the link above, but I couldn't find an issue in this repo so thought it was time to add one.

Thanks,
Ben

bb09961, 2023-10-10:
This does not solve the weather service issue, but you may want to try creating a scheduled reboot program as described here:

https://opensprinkler.com/forums/topic/opensprinkler-unified-firmware-2-1-99-released/

sumnerboy12, 2023-10-10:
I wasn't aware of that little trick, thank you!

sumnerboy12, 2024-03-11:
This is still an issue with my device, even with the nightly reboot happening, Is there anything else I can try to resolve this?

cehser, 2024-05-06:
I am experiencing the same problem. I am getting wheather service errors until I reboot the device. Then it works exactly once.

The automatic daily reboot makes it still work someway. But what can I try to find the cause of this problem? 

[image]

[image]

closed 2025-03-06 by rayshobby