OpenSprinkler/OpenSprinkler-Firmware #258: Pause behaviour does not match API docs

curlydingo, 2024-01-28:
Hi,

In adding some features to the Home Assistant integration with OpenSprinkler, we've noticed that the pause behaviour specified in the API docs does not match the firmware's behaviour. Specifically, in 23: Pause Queue:

"This command triggers a pause with the specified duration (in units of seconds). Calling it first with a non-zero duration will start the pause; calling it again with duration 0 will cancel the pause and resume station runs"

The first call works successfully, but a second call with *any* value (not just zero) will cancel the pause and resume station runs. This has become an issue for a user who was attempting to extend the duration of a pause by setting a new duration, only to have the sprinklers turn on again. Refer vinteo/hass-opensprinkler#276

I've had a look at the firmware code and it seems clear that it's designed as a 'toggle' for pause, rather than unpause-on-zero behaviour in the docs. I'm not sure what is in error here - should the firmware implement what the API docs say, or vice versa?

Using firmware version 2.2.0 and API docs for same. 

Thanks

rayshobby, 2024-01-28:
You are right that the API doc doesn't match the actual behavior. At the moment the firmware does a toggle as you described. This is possibly fairly easy to fix, basically just change the toggle_pause function here:
https://github.com/OpenSprinkler/OpenSprinkler-Firmware/blob/master/program.cpp#L128
to call resume_station only if it's in pause state AND the delay is 0. But obviously this has to be tested first. I will modify the doc to match the API behavior and perhaps fix the firmware in the next release.

rayshobby, 2024-07-25:
I've fixed the API doc to reflect the current implementation. I've added the feature request to our todo list (i.e. extend instead of toggle), and will make sure the doc matches the implementation in the future.

closed 2024-07-25 by rayshobby