OpenSprinkler/OpenSprinkler-Firmware #266: Add ability to customize Test All Stations Program

starsoccer, 2024-04-24:
Currently it seems like Test All Stations defaults to running all stations for 1 minutes which is great for basic things, but it would be great if there was a way to save a program where all stations run for say 5 minutes instead for zones that are larger and take more time to test.

rayshobby, 2024-04-28:
Please note that you can always create a program, set your desired test time, and set the program to be 'Disabled'. That way this program doesn't run normally, but you can run it manually, either from the web UI / app, or on the controller itself using buttons. This allows you to set the test program any way you want.

closed 2024-04-28 by rayshobby

starsoccer, 2024-04-28:
> Please note that you can always create a program, set your desired test time, and set the program to be 'Disabled'. That way this program doesn't run normally, but you can run it manually, either [...]

Agreed that does work, its just strange because it requires a time/date or pattern to run the program which for test ones is a bit strange. Alternatively if there was an easy way to change all station runs to be the same amount of time on a program that would be useful too.

Right now having to go one station at a time to edit 10+ zones is a pain.