OpenSprinkler/OpenSprinkler-Firmware #267: Feature Request: Local caching of weather data to allow ETO to account for days since last irrigation, instead of last day.

christianlittle7, 2024-04-30:
Currently (To my understanding) the ETO module only accounts for the previous day in it's calculations. This is fine for daily watering, and potentially every other day, but as you get into more infrequent waterings this can be less and less useful.

Having more of a full picture of the 'average' adjustment rate (The actual data doesn't even need to be kept) since the last program run could make this much much more useful.

If it rained a crazy amount the day before last, but was hot and sunny the next day, the weather adjustment might end up overwatering when it should be watering less.

mellodev, 2025-05-12:
I have the same need, multiple zones which run once weekly but I have a highly fluctuating eTO.

@christianlittle7 do you have any ideas for algorithm options to determine scaling %? Would averaging the last n days worth of eTO data be sufficient?

Dillweed04, 2025-05-16:
+1! It would be really nice to be able to customize the calcs. 

Hopefully this gets supported soon, but in the meanwhile I setup a python script to run on a rpi that pulls data from OpenWeatherMap and Open-Meteo to calculate a set number of proceeding days ETo and adds them together (I think this makes sense?....). It then sends a watering % update to Open Sprinkler through the API. It also checks for forecasted rain in the next 48 hours and will delay watering if the forecast is over a set limit. Script example output below. I'm interested in any feedback (ignore the mixed units. There is some conversions happening in the log I output.). 

Date         |  Solar (MJ/m²) |  Wind (m/s) | tmin (c) | tmax (c) | rh_min | rh_max | rain (mm) | ET₀ (in) |
------------------------------------------------------------------------------------------------------------
2025-05-15   |          22.32 |        4.19 |    19.05 |    29.84 |  51.80 |  74.00 |      0.00 |     0.26 |
2025-05-14   |          25.84 |        3.03 |    16.52 |    27.94 |  52.50 |  75.00 |      0.00 |     0.26 |
2025-05-13   |          15.71 |        2.39 |    15.50 |    22.06 |  53.20 |  76.00 |      8.48 |     0.16 |
2025-05-12   |          17.73 |        3.06 |    16.15 |    24.36 |  54.60 |  78.00 |     19.81 |     0.19 |

4 Day ET₀: 0.87 in
Total rain: 1.11 in (Effective: 0.87 in)
Net irrigation need: 0.00 in
Forecast rain next 2 days: 0.40 in
Rain is forecast — skipping watering (set scale to 0%)

Dillweed04, 2025-06-30:
@Ncp294 

Not sure if this is 100% correct, but here is what I did. 

BASELINE_ETO_IN = 0.25 #daily
DAY_COUNT = 4 #Days between watering
BASELINE_ETO_IN *= DAY_COUNT

eto_in = eto_week_mm / 25.4 #sum of calculates Eto over the days between watering
rain_in = total_rain_mm / 25.4 #sum of rain over days between watering
effective_rain = min(rain_in * 0.8, eto_in) #take 80% of rain
net_eto = max(0, eto_in - effective_rain) #account for rain to reduce watering

scale = round((net_eto / BASELINE_ETO_IN) * 100) #calc the scale. 

If everything is normal like your example I end up with 100% watering. If it is high or low the scale is adjected accordingly.