OpenSprinkler/OpenSprinkler-Firmware #268: Reboot Loop

CZonin, 2024-05-04:
I noticed that my sprinklers weren't available in the app or Home Assistant. I checked the controller and saw that it was in a reboot loop. It connects to my network > NTP Sync > Check Weather > back to WiFi logo. I've done a factory reset and still end up running into the same issue. Any ideas?

jasonhalpern, 2024-05-09:
I saw the same and have been working with the support team -- I believe they have identified a fix (mine is, as of tonight, working again). The issue appears to be related to DDoS protection in front of the weather API throwing the system off, and they're working to fine-tune it.

rayshobby, 2024-05-09:
Sorry about the inconvenience. As jasonhalpern said, this issue does not directly have to do with the firmware, it's a server side issue (though in the future we will add an option in the firmware to manually disable connection to the weather server, to avoid the reboot loop issue). I am closing this for now as we've identified what's causing the problem on the server side and will be fixing it asap.

closed 2024-05-09 by rayshobby