OpenSprinkler/OpenSprinkler-Firmware #279: Says "Stations Currently Paused" even when running a program

neilh10, 2024-07-06:
I'm enjoying testing OpenSprinkler DC version. Really great to learn it.

With my system 2.2.0 from the #sprinklers page, it says on the bottom "Stations Currently Paused" - and even when running a program - like S01 which says "Running .."
Since its running, should it say "Stations Currently Paused"
I would only expect something like this if "stop All Stations" was activated.
Here is the screen 
[image]

rayshobby, 2024-07-13:
What's the minor revision of your firmware version? 2.2.0 is the major revision number, it should also have a minor revision in the parentheses, like 2.2.0(3). At the homepage, swipe left to right to open the side menu, then click 'About', it will show the firmware version.

neilh10, 2024-07-13:
Its version 2.2.0(4)  - I built it from the github  - to do some learning :) 
 ~ oops ~
Pulling in the latest it now shows 2.2.0(5)      :)    :)     :)  
and there is some nice status while the sprinkler is running 🥇 

[image]

and also when its finished

[image]

Gosh, so sorry for getting ahead without realizing it. 

Well I'm learning (hopefully), though differencing latest with 2.2.0(3) I can't see how the firmware has changed its response to give the app the status info. !

closed 2024-07-13 by neilh10

rayshobby, 2024-07-14:
So there was a glitch in minor revision (4) -- it was released quickly to address the issues of Rasbpain Bookworm's GPIO control issue. It was basically only meant for OpenSprinkler Pi, and only shortly after that I realized in the middle of releasing it, it has introduced a bug for OS 3.x (caused by time_t being 8 bytes long which is inconsistent with unsigned long on ESP8266, which is only 4 bytes long). Long story short, the minor revision (4) was not tested on OS 3.x so if you use it you will see the station paused bug. Revision (5) as is currently in the repository should have this issue fixed.

I would generally recommend using our pre-compiled firmwares for OS 3.x:
https://openthings.freshdesk.com/support/solutions/articles/5000832310
the most recent pre-compiled firmware for OS 3.x is still 2.2.0(3), we haven't released a 2.2.0(4), again, because (4) is to address an issue for Raspbian version 12.

Hope this explains the bug you observed.

neilh10, 2024-07-15:
Thanks for the explanation.    I saw the changes to time :) and guessed there could be something happening there, so and many thanks for the explanation.  
   
I'm in learning mode, and also characterizing what works and what might be a squiggy (if anything). 
 Its been great so far, and apologies for not following the recommended approach, especially for testing, as I didn't realize what was happening. Jumping in and trying stuff is the only way I know of learning. 
I've been working with "a professional group" on some other complex software and its been challenging where they set their bar, so great to see the testing  here and everything holding up, not to mention also the forum and user support. Thanks so much.   (https://github.com/ODM2/ODM2DataSharingPortal/issues/685  
 https://www.envirodiy.org/n-ca-mayflys-through-the-winter-storms/ )