OpenSprinkler/OpenSprinkler-Firmware #284: Can't Compile ospi

domink06, 2024-07-19:
Hello,

cant compile 3ee6094 get errors:
`
ospi@ospi:~/OpenSprinkler-Firmware $ sudo ./build.sh ospi
Building OpenSprinkler...
Installing required libraries...
Hit:1 http://deb.debian.org/debian bookworm InRelease
Hit:2 http://deb.debian.org/debian-security bookworm-security InRelease
Hit:3 http://deb.debian.org/debian bookworm-updates InRelease
Hit:4 http://archive.raspberrypi.com/debian bookworm InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libmosquitto-dev is already the newest version (2.0.11-1.2+deb12u1).
raspi-gpio is already the newest version (0.20231127).
libi2c-dev is already the newest version (4.3-2+b3).
libssl-dev is already the newest version (3.0.13-1~deb12u1+rpt1).
libgpiod-dev is already the newest version (1.6.3-1+b3).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
using libgpiod
Compiling ospi firmware...
ls: cannot access 'external/TinyWebsockets/tiny_websockets_lib/src/*.cpp': No such file or directory
ls: cannot access 'external/OpenThings-Framework-Firmware-Library/*.cpp': No such file or directory
In file included from main.cpp:27:
OpenSprinkler.h:64:14: fatal error: OpenThingsFramework.h: No such file or directory
   64 |     #include "OpenThingsFramework.h"
      |              ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from OpenSprinkler.cpp:24:
OpenSprinkler.h:64:14: fatal error: OpenThingsFramework.h: No such file or directory
   64 |     #include "OpenThingsFramework.h"
      |              ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from program.h:33,
                 from program.cpp:25:
OpenSprinkler.h:64:14: fatal error: OpenThingsFramework.h: No such file or directory
   64 |     #include "OpenThingsFramework.h"
      |              ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from opensprinkler_server.cpp:25:
OpenSprinkler.h:64:14: fatal error: OpenThingsFramework.h: No such file or directory
   64 |     #include "OpenThingsFramework.h"
      |              ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from utils.cpp:26:
OpenSprinkler.h:64:14: fatal error: OpenThingsFramework.h: No such file or directory
   64 |     #include "OpenThingsFramework.h"
      |              ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from weather.cpp:25:
OpenSprinkler.h:64:14: fatal error: OpenThingsFramework.h: No such file or directory
   64 |     #include "OpenThingsFramework.h"
      |              ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from mqtt.cpp:43:
OpenSprinkler.h:64:14: fatal error: OpenThingsFramework.h: No such file or directory
   64 |     #include "OpenThingsFramework.h"
      |              ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
Done!
`

With Commit 94fa0af it works!

Greetings Dominik

arfrie22, 2024-07-19:
Hello, please run

```
git fetch
git pull --recurse-submodules
```

and then run `sudo ./build.sh ospi` again

rayshobby, 2024-07-21:
The build script has now been modified to handle submodules so I am closing this issue now.

closed 2024-07-21 by rayshobby