OpenSprinkler/OpenSprinkler-Firmware #286: segmentation fault on saving new program

finnock, 2024-07-21:
Right now im Trying to find out if OpenSprinkler would work out for me, so I booted up a Raspberry Pi 4 with a fresh install of the newest version of Raspberry Pi OS in 64bit (Debian Bookworm port). I do not have the OpenSprinkler board connected since I dont own one yet, and only want to test the software and see if it suits my needs.

I installed the OpenSprinkler software following the instructions from here:
https://opensprinklershop.de/pi/

In `sudo raspi-config` I activated the I2C Interface (although it was found under 3 -> 5 and not 3 -> 4 as stated by the instuctions)

then followed up with:

`sudo apt-get install build-essential
sudo apt-get install -y libi2c-dev 
sudo apt-get install -y libgpiod-dev
cd ~
git clone https://github.com/OpenSprinkler/OpenSprinkler-Firmware.git
cd OpenSprinkler-Firmware
sudo ./updater.sh
`

After starting the OpenSprinkler Server with `./OpenSprinkler`

it was available under `localhost:8080` as expected.

I was then able to test everything I wanted until I tried to save a new program. As soon as I hit `Save New Program` in the UI it instantly answers with "timeout" and in the console window where I started the server I see a `segmentation fault`

Is this a known problem? How can I fix it? Does this have to do with the fact that I'm running without the actual OpenSprinkler board?

rayshobby, 2024-07-21:
Sorry, we made a mistake in one commit, instead of merging master to a specific branch, we did the reverse, so the current master branch hasn't been fully tested with all API endpoints yet. I can clearly see the issue that's causing in the seg fault, but fixing all will take a bit of time. It should be done in a couple of days. In the meantime, this commit:
https://github.com/OpenSprinkler/OpenSprinkler-Firmware/commit/e0f567b3ffc389b22fc125231accc3534708cb5a
should be good to go.

If you use Raspberry Pi 5, then you will probably need to use this commit:
https://github.com/OpenSprinkler/OpenSprinkler-Firmware/commit/94fa0afcb7b96a006ed9db346a62c2914fa420a9

The main addition to the current master (which shouldn't have been merged into master in the first place) is the support for OTC for OSPi and all Linux-based platforms. In any case, sorry about the issue and it should be fixed very soon.

finnock, 2024-07-21:
thanks for the fast answer! how do I downgrade to this specific version?

STidrvr, 2024-07-22:
@rayshobby Thanks for getting my account working, and this link to the issue.

@finnock thanks for posting the above steps to revert. I have no idea what I'm doing with git's and commits and versions. I followed your steps and got mine working as well.

Thanks guys!

rayshobby, 2024-07-25:
This issue has been addressed in PR #288 which has been merged to the master branch.

closed 2024-07-25 by rayshobby