OpenSprinkler/OpenSprinkler-Firmware #289: Errore while compiling OSPI

adepiero, 2024-07-26:
```
pi@irrigazione:~/OpenSprinkler-Firmware $ git fetch
pi@irrigazione:~/OpenSprinkler-Firmware $ git pull --recurse-submodules
hint: Pulling without specifying how to reconcile divergent branches is
hint: discouraged. You can squelch this message by running one of the following
hint: commands sometime before your next pull:
hint:
hint:   git config pull.rebase false  # merge (the default strategy)
hint:   git config pull.rebase true   # rebase
hint:   git config pull.ff only       # fast-forward only
hint:
hint: You can replace "git config" with "git config --global" to set a default
hint: preference for all repositories. You can also pass --rebase, --no-rebase,
hint: or --ff-only on the command line to override the configured default per
hint: invocation.
Already up to date.
pi@irrigazione:~/OpenSprinkler-Firmware $ sudo ./build.sh ospi
Building OpenSprinkler...
A git submodule is not initialized.
Submodule 'external/OpenThings-Framework-Firmware-Library' (https://github.com/OpenThingsIO/OpenThings-Framework-Firmware-Library.git) registered for path 'external/OpenThings-Framework-Firmware-Library'
Submodule 'external/TinyWebsockets' (https://github.com/gilmaimon/TinyWebsockets.git) registered for path 'external/TinyWebsockets'
Cloning into '/home/pi/OpenSprinkler-Firmware/external/OpenThings-Framework-Firmware-Library'...
Cloning into '/home/pi/OpenSprinkler-Firmware/external/TinyWebsockets'...
Submodule path 'external/OpenThings-Framework-Firmware-Library': checked out '6ae4f2701323a1c14b63f0d22b78eaf9d6717328'
Submodule path 'external/TinyWebsockets': checked out 'a939875762af6fcca8f41a3223ca7c526d516a25'
Installing required libraries...
Hit:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Hit:2 http://archive.raspberrypi.org/debian bullseye InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libgpiod-dev is already the newest version (1.6.2-1+rpi1).
libi2c-dev is already the newest version (4.2-1+b1).
libmosquitto-dev is already the newest version (2.0.11-1+deb11u1).
libssl-dev is already the newest version (1.1.1w-0+deb11u1+rpt1).
raspi-gpio is already the newest version (0.20191001).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Compiling ospi firmware...
external/TinyWebsockets/tiny_websockets_lib/src/websockets_client.cpp: In member function ‘bool websockets::WebsocketsClient::send(const char*)’:
external/TinyWebsockets/tiny_websockets_lib/src/websockets_client.cpp:352:33: error: ‘strlen’ was not declared in this scope
  352 |         return this->send(data, strlen(data));
      |                                 ^~~~~~
external/TinyWebsockets/tiny_websockets_lib/src/websockets_client.cpp:6:1: note: ‘strlen’ is defined in header ‘<cstring>’; did you forget to ‘#include <cstring>’?
    5 | #include <tiny_websockets/internals/wscrypto/crypto.hpp>
  +++ |+#include <cstring>
    6 |
external/TinyWebsockets/tiny_websockets_lib/src/websockets_endpoint.cpp: In member function ‘bool websockets::internals::WebsocketsEndpoint::send(const char*, size_t, uint8_t, bool, bool, const char*)’:
external/TinyWebsockets/tiny_websockets_lib/src/websockets_endpoint.cpp:394:21: error: ‘memcmp’ was not declared in this scope
  394 |         if (mask && memcmp(maskingKey, __TINY_WS_INTERNAL_DEFAULT_MASK, 4) != 0) {
      |                     ^~~~~~
external/TinyWebsockets/tiny_websockets_lib/src/websockets_endpoint.cpp:2:1: note: ‘memcmp’ is defined in header ‘<cstring>’; did you forget to ‘#include <cstring>’?
    1 | #include <tiny_websockets/internals/websockets_endpoint.hpp>
  +++ |+#include <cstring>
    2 |
external/OpenThings-Framework-Firmware-Library/LinuxLocalServer.cpp: In member function ‘virtual void OTF::LinuxLocalClient::print(const char*)’:
external/OpenThings-Framework-Firmware-Library/LinuxLocalServer.cpp:53:32: error: ‘strlen’ was not declared in this scope
   53 |   client.write((uint8_t*)data, strlen(data));
      |                                ^~~~~~
external/OpenThings-Framework-Firmware-Library/LinuxLocalServer.cpp:3:1: note: ‘strlen’ is defined in header ‘<cstring>’; did you forget to ‘#include <cstring>’?
    2 | #include "LinuxLocalServer.h"
  +++ |+#include <cstring>
    3 |
^C
pi@irrigazione:~/OpenSprinkler-Firmware $
```

I am receiveing these error messages while compiling new OSPI on Raspberry PI Zero with Bullseye distribution. I aborted the process in order to save the current binaries.

margaale, 2024-07-26:
I'm getting the same errors in the exact same setup

mcastagna-sa, 2024-07-26:
Sorry, my bad. Mine is running on buster, not bullseye.

margaale, 2024-07-26:
Sorry, my bad. Mine is running on buster, not bullseye..

arfrie22, 2024-07-26:
Hello, please pull the latest release of the code and try to compile it again. It should be fixed now by #290.

margaale, 2024-07-26:
I can confirm that it has been fixed with that commit. Thanks

closed 2024-07-26 by arfrie22