OpenSprinkler/OpenSprinkler-Firmware #291: Random Disconnect from OpenThings Cloud

finnock, 2024-07-28:
My OpenSprinkler System (via OSPi) randomly disconnects from the OpenThings Cloud about once a day.

Is this a known issue? right now I have to remote into the rasp to kill the process and restart it for it to reconnect. Is there any possibility to reconnect from the UI?

rayshobby, 2024-07-29:
What's your RPi version and Raspbian version? Which firmware version? Please remember to report these as we need these in order to reproduce the issue.

finnock, 2024-07-29:
My Raspbian Version is: `Debian GNU/Linux 12 (bookworm)`
App Version: `2.4.0`
Firmware: `2.2.1`
Hardware Version: `OSPi`

It just happened again. I can still reach the OpenSprinkler in my local LAN / WiFi but via OTC its not reachable. The UI under System Diagnostics shows this:

[image]

rayshobby, 2024-07-29:
Can you go to your OTC settings, change the OTC server from ws.cloud.openthings.io to the following:
**ws.debug-cloud.openthings.io**
then reboot your system. This is the debug version of the cloud server which will provide us more information and hopefully help you figure out why yours can't connect.

finnock, 2024-07-29:
I changed the server to `ws.debug-cloud.openthings.io` and restarted the OpenSprinkler Firmware via:
`kill ####
./OpenSprinkler &`
with `####` being the `pid` of the OpenSprinkler process

what do I do next?

rayshobby, 2024-07-30:
we need to know your OTC token in order to search the debug log. For privacy reasons, you can either post the last 6 digits of your OTC token, or submit a support ticket and attach your token there.

rayshobby, 2024-08-05:
We've identified and fixed an issue that relates to OTF disconnecting / getting stuck connecting in OSPi. PR #305 has fixed this issue, and if you update the firmware you should get the latest. I'm closing this issue due to inactivity. If you observe the symptom again, you can re-open this issue.,

closed 2024-08-05 by rayshobby