OpenSprinkler/OpenSprinkler-Firmware #292: "The request was too large" error on reboot

adepiero, 2024-07-28:
OSPI compiled on Raspberry Pi Zero with bullseye.
After a system reboot I cannot connect to Opensprinkler lan server, The web page just replies "The request was too large", withouth even asking for authentication. If I restart OSPI service with `sudo systemctl restart OpenSprinkler.service`, it works.
I am connecting directly to OSPI, no proxies are involved. I receive the same error both with MS Edge and Firefox. I tried to clear the browsers' history, it doesn't work.

rayshobby, 2024-07-29:
We believe we've fixed this issue with the latest update. It's a startup script issue, as the firmware started running before the network was available. We've now changed the startup script to use systemd which allows starting the firmware after network is available. Please update the firmware and check if it works for you. Please note that we've changed the startup script from using init.d to systemd. So in case any problem comes up, just keep this in mind.

adepiero, 2024-07-29:
It works not, thank you.
The new script moved successfully my configuration from init.d to system, thank you!

psmedley, 2024-07-29:
I installed the latest firmware this morning and rebooted my pi zero. Unfortunately I still got the 'request too large' message. I'll investigate more this evening.

rayshobby, 2024-07-29:
We pushed in the fix just 1 hour ago:
https://github.com/OpenSprinkler/OpenSprinkler-Firmware/commits/master/
the fix wasn't there this morning.

psmedley, 2024-07-29:
I'm in Australia, it's 7:15am here. I pulled the code after you posted about it here.

mwerski, 2024-07-31:
Unfortunately this fix does not work for me - still the exact same symptom.

rayshobby, 2024-07-31:
Could you please report some details of your hardware? Like which RPi hardware, which Raspbian version? We need some details in order to reproduce the issue.

mwerski, 2024-07-31:
Sure! It is a model 3, connected via internal ethernet (DHCP), more precisely it is a UniPi Neuron S103 (but that should not matter in regards to network stuff) running on Debian Buster. Please drop me a note if you need more information.

`root@opensprinkler:/usr/local/OpenSprinkler-Firmware# lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 10 (buster)
Release:	10
Codename:	buster
root@opensprinkler:/usr/local/OpenSprinkler-Firmware# lshw
opensprinkler
    description: ARMv7 Processor rev 4 (v7l)
    product: Raspberry Pi 3 Model B Rev 1.2
    serial: 000000005ba6452c
    width: 32 bits
    capabilities: smp
  *-core
       description: Motherboard
       physical id: 0
     *-cpu:0
          description: CPU
          product: cpu
          physical id: 0
          bus info: cpu@0
          size: 700MHz
          capacity: 700MHz
          capabilities: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 cpufreq
     *-cpu:1
          description: CPU
          product: cpu
          physical id: 1
          bus info: cpu@1
          size: 700MHz
          capacity: 700MHz
          capabilities: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 cpufreq
     *-cpu:2
          description: CPU
          product: cpu
          physical id: 2
          bus info: cpu@2
          size: 700MHz
          capacity: 700MHz
          capabilities: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 cpufreq
     *-cpu:3
          description: CPU
          product: cpu
          physical id: 3
          bus info: cpu@3
          size: 700MHz
          capacity: 700MHz
          capabilities: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 cpufreq
     *-memory
          description: System memory
          physical id: 4
          size: 923MiB
  *-usbhost
       product: DWC OTG Controller
       vendor: Linux 5.10.103-v7+ dwc_otg_hcd
       physical id: 1
       bus info: usb@1
       logical name: usb1
       version: 5.10
       capabilities: usb-2.00
       configuration: driver=hub slots=1 speed=480Mbit/s
     *-usb
          description: USB hub
          product: SMC9514 Hub
          vendor: Standard Microsystems Corp.
          physical id: 1
          bus info: usb@1:1
          version: 2.00
[... 113 lines cut ...]

commit 453515de65b63b8e3484dd07c0870cf7204d7538
Author: Rayshobby <rayshobbyshop@gmail.com>
Date:   Fri Sep 12 10:14:42 2014 -0400

    added support to accept empty API key

commit 24bdcada10c8d16bd9aa43dbc716368ead139228
Author: Rayshobby <rayshobbyshop@gmail.com>
Date:   Thu Sep 11 19:31:15 2014 -0400

    changed station delay time to use the 8-bit comrpessed representation. firmware change only, UI will pass in/out uncompressed value (in units of seconds)

commit cdc12a67b8893d42ac5b2b4cd05298a18af636b9
Author: Rayshobby <rayshobbyshop@gmail.com>
Date:   Thu Sep 11 18:43:40 2014 -0400

    initial check in of firmware 2.1.0 Arduino code

commit 333bc85de44ae1c07302685d4ecf355ce344e26c
Author: Ray <rayshobbyshop@gmail.com>
Date:   Thu Sep 11 18:17:22 2014 -0400

    Initial commit
root@opensprinkler:/usr/local/OpenSprinkler-Firmware#`

arfrie22, 2024-07-31:
I have found where the bug was happening, the request was not handling errors properly, I have now fixed that. If this still doesn't work, please re-open this issue.

closed 2024-07-31 by arfrie22

psmedley, 2024-07-31:
Confirming this resolves the issue for me.

mwerski, 2024-07-31:
Same here, issue resolved.

PR #295 (2024-07-29): Systemd startup
fixes #292
PR #300 (2024-07-31): Request too large fix
fixes #292