OpenSprinkler/OpenSprinkler-Firmware #293: Firmware 2.2.1 rev0 fails to import saved config JSON correctly

glennschler, 2024-07-28:
I have upgraded from 2.2.0 rev1 to 2.2.1
Before I upgraded, I exported my JSON configuration. When I imported I do not see my stations. Only the station #1 is still named.

To resolve I edited my JSON configuration adding the new options that a json diff shows. Importing the edited config resolved the issue.

```json
{"options":
  {"fwire":0,
   "resv1":0,
   "resv2":0,
   "resv3":0,
   "resv4":0,
   "resv5":0,
   "resv6":0,
   "resv7":0,
   "resv8":0
  }
)
```

rayshobby, 2024-07-29:
I can't reproduce the issue. What hardware platform do you have? I am not sure why adding those new options values helped -- the firmware never requires those values, if they do not exist in the import they will simply be ignored, so I am not sure why adding those mattered. Also, I assume you made a typo: the last closing parentheses should be a closing curly bracket otherwise it would not be correct json.

glennschler, 2024-07-29:
Hardware OpenSprinkler 3.0
DC-powered

Yes I must have had a typo when I manually typed those new Options here in the issue description. When I have time I will factory reset to see  if I can reproduce the issue.

glennschler, 2024-08-01:
I am not able to reproduce this issue on a second OS 3.0 device from year 2018. I am not prepared to retest on the OS 3.2 device which experienced the original problem. I will close as not able to reproduce.

closed 2024-08-01 by glennschler