OpenSprinkler/OpenSprinkler-Firmware #296: The station automatically switches off the Tasmota connected socket about ~one minute after a startup

Tikilou, 2024-07-30:
The station automatically switches off the Tasmota connected socket about one minute after a startup from a third-party application or the socket's on/off button.

I'm encountering this with OpenSprinkler, at first I thought it was coming from my Tasmota firmware connected outlet or from MQTT, but it turns out that in the Tasmota console, the command appears to have been sent by OpenSprinkler.

[image]

[image]

[backup-complet-initial.json](https://github.com/user-attachments/files/16428925/backup-complet-initial.json)

rayshobby, 2024-07-30:
Whenever you turn on a zone, you need to provide a duration. What duration did you give to it?

Tikilou, 2024-07-30:
> Whenever you turn on a zone, you need to provide a duration. What duration did you give to it?

Nothing, because i don't always use opensprinkler for enabling water pump.
This issue appening only when i switch ON manually the tasmota socket, without opensprinkler. After that, opensprinkler send the OFF command.

Sometimes i need to enable somes without OpenSprinkler but it's a problem if he power off when i didn't asked him to power on. :-)

glennschler, 2024-07-30:
@Tikilou Why not only control on/off of the Tasmota device with the OpenSprinkler station as you show it? I use many Tasmota devices as well, though for my pump start relay I centralize control through OpenSprinkler. I have my OpenHab hub coordinate most of my house automation with MQTT subscribe and control of Tasmota devices, and also HTTP the OpenSprinker station. With the latest OS firmware I thinking to rely on MQTT even more.

rayshobby, 2024-07-30:
OK, I don't understand: when you turn on a zone in OpenSprinkler, you have to provide a duration, you can not ignore the duration. Maybe the issue is there: go to your OpenSprinkler's Edit Options -> Advanced, do you see a 'Special Station Autorefresh' check box? Is it checked? If so, you need to check it off. When that box is on, the firmware will periodically send command to the HTTP zone to make sure it's synchronized with the controller. If you also use other ways to manage your HTTP device, then you need to check that option off, so OpenSprinkler will not keep sending refreshing commands.

Tikilou, 2024-07-30:
> @Tikilou Why not only control on/off of the Tasmota device with the OpenSprinkler station as you show it? I use many Tasmota devices as well, though for my pump start relay I centralize control [...]

if it were up to me, I'd use opensprinkler and its integration with home assistant all the time.
But my wife and kids don't always agree...
It's even truer when you consider that we severely restrict the use of phones by children, and that the youngest doesn't have one.
And as it happens, they have to use this water to do their cleaning chores for the chickens and the cat's litter box.

Tikilou, 2024-07-30:
> OK, I don't understand: when you turn on a zone in OpenSprinkler, you have to provide a duration, you can not ignore the duration. Maybe the issue is there: go to your OpenSprinkler's Edit Options [...]

Thank you very much, I hadn't seen that option and indeed, OpenSprinkler now no longer turns the pump off!

rayshobby, 2024-07-30:
OK, that's good to hear. Yeah, I figured it must be that checkbox, given all the symptoms you described.

closed 2024-07-30 by rayshobby