OpenSprinkler/OpenSprinkler-Firmware #298: OpenThingsFramework is only for non- linux versions ... ie ARDUINO

booboodoguy, 2024-07-31:
https://github.com/OpenSprinkler/OpenSprinkler-Firmware/blob/52889c8fd12a57997f6ff35665717795e5270fe8/OpenSprinkler.h#L65C1-L65C34

OpenSprinkler.h
59 else // headers for RPI/BBB/LINUX
	#include <time.h>
	#include <string.h>
	#include <unistd.h>
	#include <netdb.h>
	#include <sys/stat.h>
	#include "OpenThingsFramework.h"
	#include "etherport.h"
	#include "smtp.h"
68 #endif // end of headers

rayshobby, 2024-07-31:
It compiles under both Linux and Arduino. Would you mind typing a few more words to articulate please?

closed 2024-07-31 by rayshobby

booboodoguy, 2024-07-31:
The problem occurs when compiling using vscode with the platformio extension. I removed all the env directives in the  platformio.ini except the linux directives.  The compile build fails due to the "OpenThingsFramework.h" file being not found.
The opensprinler.h file "include" causes the need for OpenThingsFramework.h.

Am I correct in assuming the "OpenThingsFramework" is not necesssary for linux.

booboodoguy, 2024-07-31:
Just looked at the build script again. I now see where the OpenThingsFramework files are located. So could it be vscode wants all the files to be in the same folder?

rayshobby, 2024-07-31:
As the instructions in the README file says:
https://github.com/OpenSprinkler/OpenSprinkler-Firmware
For OSPi and all Linux-based systems, you should follow the instructions to compile using the build.sh script. The environments in platformio is only for syntax highlighting (so that it's easier to write code in VSCode) -- the platformio is only meant for compiling Arduino firmwares, it's NOT for compiling linux.

booboodoguy, 2024-08-01:
Yes I have used the build script many times in the past. I've used MS visual studio for many years developing large complicated c++ programs. Now I would like to develop code for opensprinkler in the same type of environment using vscode and remote debugging. I have not been able to compile the code in vscode without numerous errors. Most having to do with missing files.
It's still not clear to me what the task and launch json files need to include to resolve all the external file linkages. Also, the build script requires root privileges and how to do that in vscode is not obvious.  So after all these years of opensprinkler development surely you must know how to accomplish what I am trying to do. Could you please offer some help.

rayshobby, 2024-08-01:
Honestly if I knew how to do it, I would have done it already. I am not very familiar with VS Code -- for a long time the firmware was compiled using Makefile only. It sounds like what you are trying to do is to replacing running build.sh by a button in VS Code. You are aware you can run a terminal command in VS Code as well, right?

booboodoguy, 2024-08-02:
Well you did give me a good clue about including files not in the vscode G++ compiler command. Anyway I have figured it all out and it is all working now.  I have updated my thread on the forum and  provided the final json files I use  key to getting vscode to work.  I should emphasize my goal was not to simply use vs code to replace build.sh for compiling opensprinkler. My goal was to create a professional code development environment that can be used to develop and debug new opensprinkler features. Opensprinkler is far to complicated to use print statements here and there to debug code.