OpenSprinkler/OpenSprinkler-Firmware #308: UI doesn't load if OpenSprinkler.com is down

MarcSeebold, 2024-08-13:
opensprinkler.com down -> No local UI

It seems [opensprinkler.com](http://opensprinkler.com) is down. This also affects [https://ui.opensprinkler.com/index.html](https://ui.opensprinkler.com/index.html) which makes my OpenSprinkler not loading its UI. Is there any way to decouple my OpenSprinkler from the web? It doesn't make sense to me that the UI won't load if [opensprinkler.com](http://opensprinkler.com) is unreachable.

rayshobby, 2024-08-15:
We will move ui.opensprinkler.com to a static hosting site. This is not a firmware issue.

closed 2024-08-15 by rayshobby

MarcSeebold, 2024-08-15:
I think it's a firmware issue since the firmware code uses ui.opensprinkler.com in its code. E.g., https://github.com/OpenSprinkler/OpenSprinkler-Firmware/blob/d39115ff35778d8b8befe1660cbaf89cd5fbcdff/html/sta_update.html#L60

The OpenSprinkler firmware should work without Internet access. This can be done by moving all static content to the firmware.

rayshobby, 2024-08-15:
There isn't enough flash memory space to store the static content. Did you look at how big the UI asset is? It's several mega bytes. The controller only has 2MB flash memory space in total. it's not possible to fit the entire UI asset.

markwmuller, 2024-08-19:
Can the remote data be hosted on github, for example, by default? I know that, for me, the following are true: 

- I'm willing to pay slightly more for something that is "open", and live with perhaps a few surprises where things aren't ironed out perfectly 
- One of my main motivations for buying opensprinkler, rather than one of the other "smart" sprinkler controllers, was that I wanted something that would work without relying on someone else. Buy the hardware, and forget about it. No fears about the supplier suspending support, or whatever. 
- FWIW, the opensprinkler website gave me the impression that it'd work even without internet. Admittedly, going back, sure, the information is out there, but I generally don't want to have to second guess everything I read for some technicality. 

Hosting files on something like github seems like a reasonable compromise; it speaks to the Open-ness of the project and solves the technical limitation of flash memory.

rayshobby, 2024-08-19:
Sure, if you want the js to be distributed by gh cdn, you can already do that yourself, as explained in this post:
https://www.reddit.com/r/opensprinkler/comments/1erj200/comment/li4adss/

I am not sure how hosting the JS files on a cloud server has anything to do with the project itself being Open or not. It would be akin to question the open-ness of a project if it exists on the cloud server as opposed to exist on your local computer. As I said, the UI files are already publicly available in the OpenSprinkler-App repo:
https://github.com/OpenSprinkler/OpenSprinkler-App
If you don't like it to be hosted on a cloud server, you can clone it to your computer:
https://openthings.freshdesk.com/support/solutions/articles/5000164006-customize-the-ui-app
you can also use the OpenSprinkler mobile app, which again is basically downloading the UI assets to your phone so it doesn't refer to JS files on the cloud.

markwmuller, 2024-08-19:
If I wanted to do it myself, I wouldn't have paid $150 for your product

MarcSeebold, 2024-08-19:
GitHub doesn't allow hotlinking [0]. However, [Github pages](https://pages.github.com/) might.
I think @markwmuller's issue is that the owner of [ui.opensprinkler.com](ui.opensprinkler.com) can change files so that every OpenSprinkler user is served malicious scripts. To tackle that, the firmware could use the `integrity` attribute when using `script` tags [1].

[0] https://github.blog/news-insights/product-news/heads-up-nosniff-header-support-coming-to-chrome-and-firefox/
[1] https://stackoverflow.com/questions/68487357/content-security-policy-should-a-csp-contain-hashes-for-external-scripts

salbahra, 2024-08-20:
Thank you for raising this very important issue.

First, I want to reply to some points outside the scope of this issue. The UI is very simple and does not need any bundling/building steps which means it's easy to self host if you are worried about `ui.opensprinkler.com` serving malicous content. The point about the integrity makes perfect sense however it mean's the UI version is coupled with the firmware which is a complexity we don't wish to take on at this time (each UI update triggers a new hash meaning the firmware has to update it's integrity property. Furthermore, as Ray pointed out we provide free applications on all major platforms so that the application is available in as many ways as possible.

Going back to the issue topic. You are absolutely correct we were hosting the UI assets along side our main Wordpress/Woocommerce website. As a result, when the website went down, so did the UI assets. That was not the indended behavior though as we use Cloudflare CDN in front of our web server. We have rules setup for caching but somehow they were misconfigured as it resulted in you noticing the downtime. So we have two issues, a CDN/caching issue and a coupling issue.

To resolve both of these issues, we have moved all UI (ui, app and betaui) resources to Firebase Hosting which uses Google Cloud for hosting and CDN. We have added the proper CORS headers and caching headers to ensure our resources are cached on the CDN and on the user's browser. We also employ a service worker which should protect from server issues as well. Each UI is deployed to it's own site keeping individual separation.

As a result, we believe this issue to be resolved for now. Please let us know if you have any questions or concerns.