OpenSprinkler/OpenSprinkler-Firmware #309: MQTT updates are received but control does not work after 24hours

wernerdaehn, 2024-08-16:
I have configured OpenSprinkler to publish updates and receive control commands. All works fine.
Next day I look, status updates are still received - if I start a station or a program this info gets published to MQTT - but sending commands to manually starting a station via MQTT does no longer work.
I restart OpenSprinkler and then all is back to normal until the next day.

I have no logs from OpenSprinkler and from the MQTT side all looks good.

App Version 2.4.1
Firmware 2.2.1
Hardware Version 3.3 - AC

wernerdaehn, 2024-08-19:
I do not want to commit to the 24 hours but it is recurring, reproducible.

rayshobby, 2024-08-19:
I think I figured out where the issue is. It looks like if there was a disconnect from the MQTT server, the subscribe would not initialize again, which probably caused the subscription to stop working. We can fix this in the firmware. But in the meantime, what MQTT broker are you using? Is it an internal server on your network or is it an external server? Is it fairly reliable or does it get offline / powered off periodically?

wernerdaehn, 2024-08-19:
Oh, that's brilliant that you found something despite the little information I was able to provide.
It is a Mosquitto MQTT server as container running on a Synology Diskstation in the same LAN and handling all connections from ZigBeeMQTT to Home Assistant. Point is, if there would be any issue with MQTT, the lights in my home would not respond immediately. Hence I can be confident, the MQTT server is very stable.
The OpenSprinkler is connected via WIFI however but the access point is 3m away. So I would not expect occasional network drops but cannot rule it out.
[image]

rayshobby, 2024-08-19:
I made a fix which is actually quite simple:
https://github.com/OpenSprinkler/OpenSprinkler-Firmware/commit/42e62c6b7b5f1074ab643e833bb91a6f12b72849
Before the fix, I was able to replicate the issue by stopping the MQTT broker then restarting it: the firmware stopped responding to subscribing commands since it didn't re-subscribe. After the fix, the issue is gone. Since we can't confirm if the issue on your side was caused by MQTT disconnection in the first place, I am not sure if this will definitely fix the issue for you, but at least it works in the test situation above. 

The fix is in a new branch (dev/221_1), which can download and compile. Or you can grab the current master repo, add the fix yourself and compile it. Then update the firmware on your controller.

wernerdaehn, 2024-08-21:
Working on it.

wernerdaehn, 2024-08-21:
It is installed, I started one station. Now we must wait for a day if it is working still.

wernerdaehn, 2024-08-22:
I can still turn it on/off via HomeAssistant -> MQTT, all working well.

wernerdaehn, 2024-08-23:
And I can still turn it on/off via HomeAssistant -> MQTT, all working well.

rayshobby, 2024-08-23:
OK, that's good to know. Thanks for reporting.

wernerdaehn, 2024-09-08:
Still all fine.
But one more suggestions: Update the status periodically, even if there was no change.

Scenario:
OpenSprinkler was watering the garden and sent the open/close events via MQTT.
Later the MQTT consumer (Home Assistant) was restarted. As there is no update sent - and it might take weeks until the next time the valve is used - thus it remains in its default state in HomeAssistant, which happens to be the Open icon.

closed 2025-03-06 by rayshobby