OpenSprinkler/OpenSprinkler-Firmware #311: GPIO station disabled on D1 Mini

agnoam, 2024-08-20:
I have built the firmware on D1 Mini using the provided `platformIO` configurations.
When I logged in to the UI, everything was working fine until I tried to set up a station over a specific `GPIO` pin.

How can I choose a specific GPIO?

rayshobby, 2024-08-20:
This is because there are no free GPIO pins on the official OpenSprinkler 3.x hardware. All available GPIO pins on ESP8266 have been assigned to certain functions. The free pins are defined here:
https://github.com/OpenSprinkler/OpenSprinkler-Firmware/blob/master/defines.h#L360
if you are using your own hardware and there are any spare pins, you can add them to the list, they will show up on the UI. But you will need to also change the firmware to disable the functions of these pins. For example, if you don't use any sensors, the PIN_SENSOR1 and PIN_SENSOR2 can be released as free GPIO pins, but you need to define those pins as 255 so the firmware won't initialize those pins are sensor pins.

iofirag, 2024-08-21:
> This is because there are no free GPIO pins on the official OpenSprinkler 3.x hardware. All available GPIO pins on ESP8266 have been assigned to certain functions. The free pins are defined here: https://github.com/OpenSprinkler/OpenSprinkler-Firmware/blob/master/defines.h#L360 if you are using your own hardware and there are any spare pins, you can add them to the list, they will show up on the UI. But you will need to also change the firmware to disable the functions of these pins. For example, if you don't use any sensors, the PIN_SENSOR1 and PIN_SENSOR2 can be released as free GPIO pins, but you need to define those pins as 255 so the firmware won't initialize those pins as sensor pins.

" you can add them to the list"
is there an option to leave the sensor pins and add more pin variable to this list to handle my GPIO?

rayshobby, 2024-08-22:
The PIN_FREE_LIST
I don't understand how you can use the same GPIO pin for multiple purposes? Are you asking how to use a pin both as a sensor pin as well as a free pin for GPIO station? That's not possible.

closed 2024-08-22 by rayshobby

agnoam, 2024-08-24:
@rayshobby, can you give me a reference to the place to disable the flow sensors for the `ESP8266`?

rayshobby, 2024-08-24:
How many GPIO zones are you looking for? If you needs one, I suggest you start with the RFTX (radio-frequency transmitter) pin, assuming you are not using a RF transmitter:
https://github.com/OpenSprinkler/OpenSprinkler-Firmware/blob/master/defines.h#L418
change it to 255, so that effectively disables RFTX. That can free up pin 15, and you can add it to the PIN_FREE_LIST:
https://github.com/OpenSprinkler/OpenSprinkler-Firmware/blob/master/defines.h#L476
i.e. `#define PIN_FREE_LIST {15}`
This way GPIO 15 can be used for GPIO station.

iofirag, 2026-11-09:
hi @rayshobby 
i noticed for your hard work on last version, 221(7)
and see some structure changes in `defines.h`.
how to handle it in version 221(7)?

just to remember:
in version 221(6) you told me to do this:

> How many GPIO zones are you looking for? If you need one, I suggest you start with the RFTX (radio-frequency transmitter) pin, assuming you are not using a RF transmitter: https://github.com/OpenSprinkler/OpenSprinkler-Firmware/blob/master/defines.h#L418 change it to 255, so that effectively disables RFTX. That can free up pin 15, and you can add it to the PIN_FREE_LIST: https://github.com/OpenSprinkler/OpenSprinkler-Firmware/blob/master/defines.h#L360 i.e. `#define PIN_FREE_LIST {15}` This way GPIO 15 can be used for GPIO station.

so this changes make this to work:

> just for organize. and make it readable for future.
> 
> success build it for D1 mini board (8266 included)
> 
> changes:
> 
> in `defines.h` :
> 
> find this section: `#elif defined(ESP8266) // for ESP8266`
> 
> disabled 2 sensors by set them as 255 (as you saied)
> 
> ```
> #define V0_PIN_SENSOR1       255 // sensor 1 disabled; GPIO12 is free
> #define V0_PIN_SENSOR2       255 // sensor 2 disabled; GPIO13 is free
> ```
> 
> and add them to the free pin list: 
>
> ```
> #define PIN_FREE_LIST {12,13}
> ```
> then PlatformIO build.
> 
> optional build errors:
> 
> 1. missing npm package `html-minifier-terser`?:
>    install it manual: `npm i html-minifier-terser`
> 2. many cpp, .h missing files?
>    build it again.
> 
> created a pull request just for storing the changes [#411](https://github.com/OpenSprinkler/OpenSprinkler-Firmware/pull/411)
> 
> it is available in my repo: https://github.com/aghaihome/D1-mini-Board-OpenSprinkler-Firmware/tree/master

rayshobby, 2026-11-11:
I suggest that you use an AI coding assistant (Codex, Claude Code etc.) to help. Just describe what you want to achieve and it can complete your request in a few minutes. It can also compile and upload the firmware to your board automatically.