OpenSprinkler/OpenSprinkler-Firmware #312: Local time without internet connection

iofirag, 2024-08-21:
Hi,
i have been install Opensprinkler on raspberry pi (rpi3b+) via the build.sh file,
and want Opensprinkler will use RTC module when there isnt internet connection
how can i achieve that?

rayshobby, 2024-08-22:
The microcontroller-based OpenSprinkler (i.e. OpenSprinkler 3.x) has built-in RTC which can maintain time when power is off. For RPi-based OpenSprinkler, because time keeping as well as NTP are managed by the Raspbian operating system, the firmware doesn't handle either of them. You can google 'rpi real-time clock' to install a RTC module yourself.

iofirag, 2024-08-22:
Great. Thanks.
What about `d1 mini`?

rayshobby, 2024-08-23:
If you have an RTC module connected to D1 mini (the RTC can be PCF8563 or DS1307, both very common and popular), then the firmware can auto-detect these and use them.

closed 2024-08-23 by iofirag