OpenSprinkler/OpenSprinkler-Firmware #315: UI removed `%20` encoded spaced in Main Station's configuration

deveth0, 2024-08-28:
When setting up a main station to use http as type and the on/off command `cm?cmnd=Power%20On` (as documented here: https://opensprinklershop.de/en/2020/11/21/mit-wlan-steckdose-pumpe-steuern/?v=3a52f3c22ed6), the ui (or backend) removes the escaping and saves the following:

```
"special": {
        "0": {
            "st": 4,
            "sd": "192.168.88.243,80,cm?cmnd=Power On,cm?cmnd=Power Off"
        }
    }
```

This breaks the functionallity as those commands are not send correctly then.

rayshobby, 2024-08-28:
Commenting out urlDecode is the fix. It's now in the dev221_1 branch, which is not officially released yet but will be in a few weeks:
https://github.com/OpenSprinkler/OpenSprinkler-Firmware/commit/27c7a934ff7462cd18ece177350db4b82bbc2255

deveth0, 2024-08-28:
ahhh great, so this does not have any side effects I need to worry about?

Thanks for your fast reply here!

rayshobby, 2024-08-28:
What side effect are you referring to? The background of the issue is as follows: prior to the current firmware, OSPi didn't support OTF, and the firmware perform urlDecode separately for OSPi. The current firmware supports OTF, and because OTF library already performs urlDecode, it resulted in performing it twice, which is why %20 got converted further to a space character. So the fix is to remove the unnecessary urlDecode.

deveth0, 2024-08-28:
I was a bit nervous as you mentioned [here](https://github.com/OpenSprinkler/OpenSprinkler-Firmware/issues/294#issuecomment-2256343054) that you are not fully aware of the potential side effects on other users. 

But given that you already did the change on the dev branch, I'm feeling good to use it and think this one can be closed.

closed 2024-08-28 by deveth0