OpenSprinkler/OpenSprinkler-Firmware #316: Allow to manually turn one the master zone

wernerdaehn, 2024-09-08:
I have many valves but also some manual water taps. To wash the car, fill a bucket, water plants manually.
Because all zones should turn on the pump, the zone S1 is a master. That works fine.

It is just inconsistent that I have to turn on a not-used zone to start the pump instead of simply starting it, starting the master zone all by itself.

rayshobby, 2024-09-08:
This is due to the way th scheduler is implemented. Master zone does not appear in the runtime queue, instead its status is dynamically determined by other zones. If it was an independent zone you will get into ambiguous situations where a zone that activates the master is on, but if you then tell the master itself to turn off, is it supposed to be off or on? 

Please note that the dummy zone used to activate master doesn't need to be an actual physical zone. You can define as many zones as you want, even if you don't have that many physical zones. So unless you have exhausted all the 72 available zones, you can just have a spare zone to activate the master.

wernerdaehn, 2024-09-08:
That's what I did and it works. I also understand the technical reason behind it, but as a user I have a different expectation. That's the point I am trying to make.

rayshobby, 2024-09-08:
I understand that, but have you thought about the confusion by allowing the user to operate the master zone independently? For example, if someone tells the master to turn off, I would assume that the master needs to be off. But if you have another zone that activates the master that's on, then by default the master zone cannot turn off. So you would get in a situation where the user tells the master to turn off but it cannot. That is confusing.

wernerdaehn, 2024-09-08:
True.

closed 2024-09-08 by rayshobby