OpenSprinkler/OpenSprinkler-Firmware #320: DHCP issue loss of connectivity

phoenix-9, 2024-09-24:
Firmware: 2.2.1
Hardware Version: 3.2 - AC

I am using the wired Ethernet module.  Sometimes opensprinkler loses connectivity.  The lawn still gets watered.  I cannot log on.  I cannot ping.  I look in the DHCP server logs and I see an endless DHCPDISCOVER/DHCPOFFER loop every 60 seconds.  I will post more details when it happens again and I can run tcpdump or wireshark.

Any suggestions?

skellycode, 2024-09-29:
Same issue here.  Happens every couple of days. (Firmware: 2.2.1, Hardware: 3.0 - AC connected via WiFi)

phoenix-9, 2024-11-02:
It's happened a few more times.  These are the DHCP packets that get repeated every 60 seconds.  They look unremarkable:
```
05:33:30.770252 IP (tos 0x0, ttl 255, id 49189, offset 0, flags [none], proto UDP (17), length 336)
    0.0.0.0.68 > 255.255.255.255.67: BOOTP/DHCP, Request from ac:0b:fb:f7:c9:3e, length 308, xid 0x197e85b2, Flags [none]
          Client-Ethernet-Address ac:0b:fb:f7:c9:3e
          Vendor-rfc1048 Extensions
            Magic Cookie 0x63825363
            DHCP-Message (53), length 1: Discover
            MSZ (57), length 2: 1500
            Parameter-Request (55), length 5: 
              Subnet-Mask (1), Default-Gateway (3), BR (28), Domain-Name-Server (6)
              NTP (42)
            Client-ID (61), length 7: ether ac:0b:fb:f7:c9:3e
05:33:30.771058 IP (tos 0x10, ttl 128, id 0, offset 0, flags [none], proto UDP (17), length 328)
    192.168.17.1.67 > 192.168.17.209.68: BOOTP/DHCP, Reply, length 300, xid 0x197e85b2, Flags [none]
          Your-IP 192.168.17.209
          Client-Ethernet-Address ac:0b:fb:f7:c9:3e
          Vendor-rfc1048 Extensions
            Magic Cookie 0x63825363
            DHCP-Message (53), length 1: Offer
            Server-ID (54), length 4: 192.168.17.1
            Lease-Time (51), length 4: 43200
            Subnet-Mask (1), length 4: 255.255.255.0
            Default-Gateway (3), length 4: 192.168.17.1
            BR (28), length 4: 192.168.17.255
            Domain-Name-Server (6), length 8: 192.168.17.1,192.168.17.4
            NTP (42), length 4: 192.168.17.7
```

The only additional piece of information I have is that Opensprinkler assigned itself an automatic private IP address.  It tries to query DNS (because those are the IP addresses that it's trying to connect to by looking at the ARP packets:

```
05:52:01.061096 ARP, Ethernet (len 6), IPv4 (len 4), Request who-has 192.168.17.1 tell 169.254.63.201, length 50
05:52:01.061096 ARP, Ethernet (len 6), IPv4 (len 4), Request who-has 192.168.17.4 tell 169.254.63.201, length 50
05:52:01.061181 ARP, Ethernet (len 6), IPv4 (len 4), Reply 192.168.17.1 is-at 00:08:a2:0b:c8:fe, length 28
```

closed 2025-03-06 by rayshobby