OpenSprinkler/OpenSprinkler-Firmware #322: Firmware Update Build Errors

Mattat01, 2024-10-30:
I'm confused. I'm trying to update the firmware on my OSPi, and get the following response to sudo ./build.sh ospi:
```
Building OpenSprinkler...
A git submodule is not initialized.
Submodule 'external/OpenThings-Framework-Firmware-Library' (https://github.com/OpenThingsIO/OpenThings-Framework-Firmware-Library.git) registered for path 'external/OpenThings-Framework-Firmware-Library'
Submodule 'external/TinyWebsockets' (https://github.com/gilmaimon/TinyWebsockets.git) registered for path 'external/TinyWebsockets'
Cloning into '/home/pi/OpenSprinkler-Firmware/external/OpenThings-Framework-Firmware-Library'...
Cloning into '/home/pi/OpenSprinkler-Firmware/external/TinyWebsockets'...
Submodule path 'external/OpenThings-Framework-Firmware-Library': checked out '0f74fa747ea98451ab6956be9555227d467b568d'
Submodule path 'external/TinyWebsockets': checked out 'a939875762af6fcca8f41a3223ca7c526d516a25'
Installing required libraries...
Ign:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Get:2 http://archive.raspberrypi.org/debian stretch InRelease [41.9 kB]
Err:3 http://raspbian.raspberrypi.org/raspbian stretch Release
  404  Not Found [IP: 93.93.128.193 80]
Get:4 http://archive.raspberrypi.org/debian stretch/main armhf Packages [192 kB]
Get:5 http://archive.raspberrypi.org/debian stretch/ui armhf Packages [44.6 kB]
Reading package lists... Done
E: The repository 'http://raspbian.raspberrypi.org/raspbian stretch Release' does no longer have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```
Ultimately my app still says I'm on Firmware 2.1.8(2) , and I cannot work out if the above contains critical errors that might cause this or, certainly, how to fix it.

Any assistance would be appreciated.
Thanks,
Matthew

rayshobby, 2024-10-31:
It looks like you have Raspbian Stretch, which was from 7 years ago. Most likely you need to upgrade your Raspbian to a more recent version, such as Bookworm or Bullseye

Mattat01, 2024-10-31:
Whatever I have would certainly be from 7+ years ago, which also means that I have forgotten everything about how I set it up...

Would you agree that I follow the steps to upgrade to Bullseye here: https://www.makeuseof.com/tag/raspberry-pi-update-raspbian-os/ (except target the Bullseye repo instead of the Buster (I can probably find an updated guide)), and then redo the OSPi update, or should I be reflashing an SD card and starting from scratch?

I'm comfortable updating, I'm just not familiar with the history of what changed and why over the last 7 years, so I'm not sure if it is going to be that simple! If starting from scratch is just flashing a new Raspberry Pi firmware and installing OpenSprinkler, then reuploading my config backup, that may be a less error prone way to proceed. Do you have experience with either, and/or an opinon?

Thanks

Mattat01, 2024-10-31:
I just made up a new SD with bookworm, and it is now up and running with Firmware 2.2.1.

Thanks.

closed 2024-10-31 by Mattat01