OpenSprinkler/OpenSprinkler-Firmware #327: Unable to compile for OS2.3 (pio run -e sanguino_atmega1284p)

denisfrench, 2025-01-05:
Hi; I note that the release notes for 2.2.1(0) states that firmware builds are available on GitHub upon release, but I'm using OS2.3DC and can only see a file for ESP8266 on the releases page.

So I tried compiling on a GitHub workspace. As per https://openthings.freshdesk.com/support/solutions/articles/5000165132-compile-the-opensprinkler-firmware all I had to do was add the PlatformIO IDE extension. Pressing the ```PlatformIO: Build``` button successfully compiled the ESP8266 firmware, but the AVR firmware failed.

Output from ```pio run -e sanguino_atmega1284p``` is as follows:
```
Processing sanguino_atmega1284p (platform: atmelavr; board: ATmega1284P; framework: arduino)
-------------------------------------------------------------------------------------------------------------
Library Manager: Installing greiman/SdFat @ 1.0.7
UnknownPackageError: Could not find the package with 'greiman/SdFat @ 1.0.7' requirements for your system 'linux_x86_64'
```

As an experiment I tried removing the fixed version of SdFat within platformio.ini; it did successfully download all dependencies, however then failed with a long list of compilation errors.

Thanks for your ongoing development of this firmware. This is why I use open software and hardware whenever possible 🙂

rayshobby, 2025-01-05:
Hi, first of all, please note that we release all compiled versions of the firmware on our own site:
https://raysfiles.com/os_compiled_firmware/

Second, I just ran VSCode and compiled the firmware for atmega1284p, it went through just fine, no errors. I've never run pio in command line. Have you tried VSCode, as the compilation instructions describe?

denisfrench, 2025-01-07:
Sorry; I did miss the hyperlink to the files within the firmware update guide.

I don't run a local dev environment, instead invoking a GitHub Codespace from whatever repo I want to dabble in. GitHub Codespaces include VSCode by default, presenting a web interface of VSCode running on their VM. Launching a Codespace from https://github.com/OpenSprinkler/OpenSprinkler-Firmware, I add the Platformio extension, and hit the ```PlatformIO: Build``` button. This successfully builds the d1_mini environment firmware, but not the sanguino_atmega1284p environment.

Running ```pio run -e sanguino_atmega1284p``` was just to isolate the problem, to only build the AVR firmware. The original error finding the greiman/SdFat library is the same regardless of hitting the ```PlatformIO: Build``` button, or running pio on the command line.

Upon further investigation, it seems the PlatformIO registry is not aware of a 1.0.7 version of the SdFat library, which might explain the download error. See https://registry.platformio.org/libraries/greiman/SdFat/versions . Could it be that because v1.0.7 is already on your system PlatformIO is not attempting to download it?

So I tried instead pointing directly to the 1.0.7 tagged version of SdFat on GitHub, so that section of platformio.ini looks like this:
```
[env:sanguino_atmega1284p]
platform = atmelavr
board = ATmega1284P
board_build.f_cpu = 16000000L
board_build.variant = sanguino
framework = arduino
lib_ldf_mode = deep
lib_deps =
    https://github.com/UIPEthernet/UIPEthernet/archive/refs/tags/v2.0.12.zip
    knolleary/PubSubClient @ ^2.8
    https://github.com/greiman/SdFat.git#1.0.7
;    greiman/SdFat @ 1.0.7
    Wire
build_src_filter = +<*> -<html/*> --<external/*>
monitor_speed=115200
```

This successfully downloads v1.0.7 of SdFat, but again results in a long list of compilation errors (regardless of how compiling the AVR firmware is invoked).

So if you can't reproduce this, I'm guessing there is something missing from the GitHub Codespace, or some other configuration difference.

Feel free to close this issue, given the latest firmware is in fact available on your website. I note the original error is the same as this guy was having (even if he was actually after OS3.x firmware); https://opensprinkler.com/forums/topic/announcing-opensprinkler-unified-firmware-2-2-10/#post-79662

rayshobby, 2025-01-08:
I use Linux Mint 21.3 64-bit and I cleared the whole project and rebuilt. It was still successful, I didn't encounter any problem. Instead of specifying the library the way it is, you can also directly give it the link to the zip file. For example, you can change it to:
https://github.com/greiman/SdFat/archive/refs/tags/1.0.7.zip
I just tried and it still works. You can give it a try. I am closing this issue for now since I can't reproduce it. But we can certainly change it to the zip file link if that works for you.

closed 2025-01-08 by rayshobby