OpenSprinkler/OpenSprinkler-Firmware #332: Compilation errorgpio.cpp

nriel, 2025-02-09:
Hi there,

I am not able to compile the firmware on my Rapsberry Pi. Can anyone help please?

`Compiling ospi firmware...
gpio.cpp: In function ‘void pinMode(int, unsigned char)’:
gpio.cpp:581:67: error: ‘GPIOD_LINE_REQUEST_FLAG_BIAS_PULL_UP’ was not declared in this scope
    gpiod_line_request_input_flags(gpio_lines[pin], gpio_consumer, GPIOD_LINE_REQUEST_FLAG_BIAS_PULL_UP);
                                                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
gpio.cpp:581:67: note: suggested alternative: ‘GPIOD_LINE_REQUEST_FLAG_ACTIVE_LOW’
    gpiod_line_request_input_flags(gpio_lines[pin], gpio_consumer, GPIOD_LINE_REQUEST_FLAG_BIAS_PULL_UP);
                                                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                                                   GPIOD_LINE_REQUEST_FLAG_ACTIVE_LOW
`

rayshobby, 2025-02-09:
Please report your Raspbian version. It's likely your Raspbian is outdated resulting in this error.

nriel, 2025-02-09:
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

nriel, 2025-02-09:
I did what the error said and compilation went through and Opensprinkler seems to work:
replace GPIOD_LINE_REQUEST_FLAG_BIAS_PULL_UP by GPIOD_LINE_REQUEST_FLAG_ACTIVE_LOW

But I do not know about any side effects?

rayshobby, 2025-02-09:
I am pretty sure it's because your GPIOD version is too old, and that in turn is likely because your Raspbian is an older version (the current version of Raspbian is 12 (bookworm)). If you can upgrade your GPIOD to version 1.6 that should fix the issue, otherwise you should upgrade your Raspbian.

This macro seems to affect pins which are set for internal pull-ups. Without it, you will likely encounter problems on the sensor pins and buttons. However, if you don't use any sensors or buttons, then it should be ok.

nriel, 2025-02-10:
Just checked and GPIO version seems to be up to date:

sudo apt-get -y install python-rpi.gpio
Reading package lists... Done
Building dependency tree       
Reading state information... Done
python-rpi.gpio is already the newest version (0.7.0-0.1~bpo10+4).

As I do not use any of the sensor pins nor buttons, I do not really care right now :)

rayshobby, 2025-02-10:
I was referring to the libgpiod, not python gpio, because the firmware is not written in python, it's written in C++. Since you don't use sensors or buttons, I don't think you need to worry about it for now.

closed 2025-02-10 by rayshobby