OpenSprinkler/OpenSprinkler-Firmware #334: Cannot compile 2.2.1(1)

margaale, 2025-03-06:
Hi, im running opensprinkler in a raspberry pi. I was trying to update to the latest firmware but when i try to compile it i'm getting the following error.

[image]

Let me know if you need more info or anything...

rayshobby, 2025-03-06:
What Raspbian version are you on? I believe it needs to be at least Bullseye, or Bookworm ideally.

margaale, 2025-03-06:
I'm on buster. So, latest firmrware is using some dependency that is not up to date in buster?

rayshobby, 2025-03-06:
Yes, it uses gpiod

margaale, 2025-03-06:
Oh, ok. Im stuck at 2.2.1(0) then.. Thanks

closed 2025-03-06 by rayshobby

margaale, 2025-03-06:
Just adding more info. These are the version that got installed when compiling.

[image]

rayshobby, 2025-03-06:
While gpiod exists for Buster, the version is just not up to date in order to enable GPIO pull-up. We had to switch to gpiod because RPi 5 no longer supports the older sysfs way of GPIO control. And it would be too cumbersome to have to maintain 2 different versions of the gpio functions in order to handle both Buster and Bookworm. If you can upgrade to at least Bullseye then it should work.

rayshobby, 2025-03-07:
I should mention this as well: note that the firmware can now run in Docker, you can follow the instructions here to compile it in Docker:
https://github.com/OpenSprinkler/OpenSprinkler-Firmware/blob/master/README_Docker.md
it's configured to use Bookworm so while your host system is Buster it can still compile and run just fine.

margaale, 2025-03-07:
Yes, thanks. I had seen that. But I'm running open sprinkler on a pi zero like for 5 years I think I hasn't got the necessary power to run docker...

rayshobby, 2025-03-08:
I had some more time this weekend to take a closer look at this, and I was able to get the latest firmware compile and run fine on Raspbian Buster. This was done by compiling and installing libgpiod v1.6 on Buster (which was what chatgpt suggested when I gave it the error message). To see the instructions, just go to chatgpt, and ask "how to install libgpiod v1.6 on raspbian buster". Also remember to remove the `libgpiod-dev and gpiod` from the `apt-get install` in `build.sh`, otherwise the build script will reinstall the outdated gpiod again. It involves a bunch of commands and isn't the most straightforward, but it did succeed in the end. That said, it's a good idea to upgrade to Bullseye, given that Buster is more than 5 years old now.