OpenSprinkler/OpenSprinkler-Firmware #337: no support of OSBO in 2.2.1(1)

joe-ben-75, 2025-03-23:
After a recent update from 2.2.1(0) to 2.2.1(1) I had to recognize that OSBO (BeagleBone Black configuration) is no longer supported ?!?
The define.h file does not include OSBO definitions and the build file only builds OSPI versions.
Is that intended or just a mistake ?
The Change Log does not include any hint of removing OSBO.
Would be very happy to see OSBO support again (I tried adding the OSBO parts in define.h and build.sh, but build process fails).

rayshobby, 2025-03-23:
Hi, sorry about the omission in the release notes. It's intended -- OSBo has been discontinued for more than a decade, we no longer test new firmwares with BeagleBone anymore, and don't want to release firmwares for it since it's untested.

closed 2025-03-23 by rayshobby

joe-ben-75, 2025-03-23:
Ray, thanks for the quick answer.
My OpenSprinkler controls 18 Channels, which was only possible on OSPI (which can max support 18 free GPIOs). I wanted to change to OSBO because it supports even more free GPIOs.
I tested 2.2.1(0), which worked as far I could test and wanted to continue testing now.
Unfortunately I made the mistake to update when I saw that 2.2.1(1) was available - and then I was in trouble ...
I understand the discontinuation of OSBO, but I am now not able to return to 2.2.1(0) again, which would enable me continuing my testing.
When I clone the git tag 221(0) everything looks fine until I start to build which fails due to some errors in TinyWebsockets code (I tried also OSPI, which didn't work either). But I don't understand these errors because there should be no changes in this code from 221(0) to 221(1) ?

Although OSBO is not supported anymore, could you possibly just help to get 221(0) running again ?

attached the output of the build:

opensprinkler@BeagleBone:~/OpenSprinkler-Firmware$ sudo ./build.sh osbo
Building OpenSprinkler...
Updating submodules.
Installing required libraries...
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libmosquitto-dev is already the newest version (2.0.11-1+deb11u2).
libssl-dev is already the newest version (1.1.1w-0+deb11u2).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Compiling osbo firmware...
external/TinyWebsockets/tiny_websockets_lib/src/websockets_client.cpp: In member function 'bool websockets::WebsocketsClient::send(const char*)':
external/TinyWebsockets/tiny_websockets_lib/src/websockets_client.cpp:352:33: error: 'strlen' was not declared in this scope
  352 |         return this->send(data, strlen(data));
      |                                 ^~~~~~
external/TinyWebsockets/tiny_websockets_lib/src/websockets_client.cpp:6:1: note: 'strlen' is defined in header '<cstring>'; did you forget to '#include <cstring>'?
    5 | #include <tiny_websockets/internals/wscrypto/crypto.hpp>
  +++ |+#include <cstring>
    6 | 
external/TinyWebsockets/tiny_websockets_lib/src/websockets_endpoint.cpp: In member function 'bool websockets::internals::WebsocketsEndpoint::send(const char*, size_t, uint8_t, bool, bool, const char*)':
external/TinyWebsockets/tiny_websockets_lib/src/websockets_endpoint.cpp:394:21: error: 'memcmp' was not declared in this scope
  394 |         if (mask && memcmp(maskingKey, __TINY_WS_INTERNAL_DEFAULT_MASK, 4) != 0) {
      |                     ^~~~~~
external/TinyWebsockets/tiny_websockets_lib/src/websockets_endpoint.cpp:2:1: note: 'memcmp' is defined in header '<cstring>'; did you forget to '#include <cstring>'?
    1 | #include <tiny_websockets/internals/websockets_endpoint.hpp>
  +++ |+#include <cstring>
    2 | 
external/OpenThings-Framework-Firmware-Library/LinuxLocalServer.cpp: In member function 'virtual void OTF::LinuxLocalClient::print(const char*)':
external/OpenThings-Framework-Firmware-Library/LinuxLocalServer.cpp:53:32: error: 'strlen' was not declared in this scope
   53 |   client.write((uint8_t*)data, strlen(data));
      |                                ^~~~~~
external/OpenThings-Framework-Firmware-Library/LinuxLocalServer.cpp:3:1: note: 'strlen' is defined in header '<cstring>'; did you forget to '#include <cstring>'?
    2 | #include "LinuxLocalServer.h"
  +++ |+#include <cstring>
    3 | 
external/OpenThings-Framework-Firmware-Library/Websocket.cpp: In member function 'bool WebsocketClient::send(const char*, size_t, bool)':
external/OpenThings-Framework-Firmware-Library/Websocket.cpp:189:18: error: 'strlen' was not declared in this scope
  189 |         length = strlen(payload);
      |                  ^~~~~~
external/OpenThings-Framework-Firmware-Library/Websocket.cpp:2:1: note: 'strlen' is defined in header '<cstring>'; did you forget to '#include <cstring>'?
    1 | #include "Websocket.h"
  +++ |+#include <cstring>
    2 |

rayshobby, 2025-03-23:
These are pretty basic errors indicating it's not finding string.h. You can easily find it out by simply pasting the error code to chatgpt and ask it how to fix the error. In any case, note that in the current build.sh, there are compiler flags like:
`-include string.h`
you can similarly add this to the line that compiles the firmware for OSBo and that should fix the problem. If not, you can also manually add `#include <string.h>` to the files missing this header, that should also fix the issue.