OpenSprinkler/OpenSprinkler-Firmware #338: Feature request - no flow alert

booboodoguy, 2025-03-26:
I have been using the flow rate sensor code for a couple of years now and have experienced zero flow situations due to a defective rain sensor. Now that  the flow sensor code from bb09961 is in the main branch I request that a check be added to the flow sensor flow rate to, in addition to exceeding the user specified flow rate, a flow rate greater than zero ( or some small number) is checked.

bb09961, 2025-03-30:
Yeah, that is a good idea.   I had the same thing happen last year due to a flow meter wiring issue.   To make it more flexible, it would be great to implement a user set minimum flow rate for each station.    If the flow sensor or a valve fails, or if a sprinkler head or drip system is clogged, or a master pump fails, or the water is unexpectedly shut off, it would send a **Low Flow** alert.    

If the existing method is used, it would get a little crowded and maybe confusing to add 2 numbers to the name of each station to indicate minimum and maximum flow rate.   It's possible, but I think changing the UI to have the min and max flow value stored for each station in separate fields would be ideal.   Let's see if that is something they would consider first as I don't have the ability to suggest specific UI code changes myself.

As an example, this below would start impacting the amount of room for the station name:

[image]

This would be much more intuitive:

[image]

booboodoguy, 2025-04-23:
Here's my simple code changes to notifier.cpp to get a FLOW ALERT if the flow for any zone is less than 10% of the max flow set point.

281 if (flow_last_gpm > 0 && strlen(tmp_station_name) > 5) {
remove "flow_last_gpm > 0 &&" since I want to check for zero flow.
next
293 if ((flow_last_gpm*flowrate100/100.f) > flow_gpm_alert_setpoint) {
change line 293 to ...

293 if ((flow_last_gpm*flowrate100/100.f) > flow_gpm_alert_setpoint || (flow_last_gpm*flowrate100/100.f) < (.1 * flow_gpm_alert_setpoint )) {

This solves my problem.

bb09961, 2025-04-23:
Yeah, that makes sense. I'll test that out on my personal firmware version and see how it goes.

booboodoguy, 2025-05-04:
There is a possibility the rain sensor could be active causing a low flow alert. I had an occurrence where the rain sensor became active at the same time as a program zone became active causing a flow alert message to be sent. Therefore, line 293 needs to include a rain sensor check. The following code is a better solution.

#define FLOW_SCALE 100.0f
#define LOW_FLOW_THRESHOLD 0.1f

float flow_rate = flow_last_gpmflowrate100 / FLOW_SCALE;

if (flow_rate > flow_gpm_alert_setpoint || 
    (flow_rate < (LOW_FLOW_THRESHOLD * flow_gpm_alert_setpoint) && ! os.status.sensor2_active)) {
}

bb09961, 2025-05-08:
The only issue I found with this logic is that if the station runs less than ~90 seconds, the flow rate will always be 0.  Flow rates aren't calculated before 90 seconds to avoid air in the lines causing inaccurate flow rates.  
With that in mind, I added an additional check to only send an alert if the duration is greater or equal than 120 seconds
if (fval >= 120 && ( (flow_last_gpm * flowrate100 / 100.f) > flow_gpm_alert_setpoint || (flow_last_gpm * flowrate100 / 100.f) < (0.1 * flow_gpm_alert_setpoint) ))

booboodoguy, 2025-11-16:
Correct me I am wrong but arfrie22 "Fix real time flow to use a running average with dynamic timeout #355" doesn't fix no message on zero flow #338  since....

notifier.cpp line 285 still has 	"if (flow_last_gpm > 0 && strlen(tmp_station_name) > 5) {"

anun333, 2026-10-12:
Some arithmetic for the minimum flow per station suggested above. It might help choose the
default.

A zone's flow is the sum of its heads'. Take a lawn zone of six 15-foot half heads and two
quarters (Rain Bird 15 Series MPR at 30 psi: 1.85 and 0.92 gpm). It should run at
**12.94 gpm**.

* One half head fully clogged takes it to 11.09, **−14%**.
* One quarter head, −7%.
* A rotor zone of four 5000s with #3.0 nozzles at 45 psi (3.09 gpm each) loses **25%** for
  one clogged rotor.
* A drip zone of a hundred 1-gph emitters loses 1% for one emitter. It only shows as drift
  over weeks.

Against that: the meter's own error is about ±2% (Hunter HC). A nozzle's flow goes as √P,
so a ±10% pressure swing moves the whole zone about ±5%.

* **A minimum near 90% of the zone's usual flow** catches a clogged half head or rotor. A
  quarter head sits at the edge.
* **10% of the high set point** catches only a dead line or a dead sensor, which is the
  original report and still worth having.
* **The 120-second floor** from the comment above keeps the first 90 seconds of air out of
  it.

A default that needs no typing: 90% of the median of the zone's last few runs. A number
typed in by the user would override it.

If a PR along these lines would be welcome, I'm happy to write it. The full workings are in
https://github.com/anun333/fieldmesh/blob/main/docs/notes/water-meters.md.