OpenSprinkler/OpenSprinkler-Firmware #339: mqtt error messages - Client received (but ignored)

webdandy, 2025-03-30:
Hi,
after upgrading to 2.2.1(1) I‘m receiving error messages on my mqtt broker (iobroker with mosquitto):

mqtt.0 2025-03-30 21:21:03.646 debug Client received (but ignored) "opensprinkler/sensor1" (string): {"state":1}

Any ideas how to fix that?

Thanks and regards
Fabian

rayshobby, 2025-03-31:
This seems to be due to how your mqtt broker was set up. I don't see anything obviously wrong with the mqtt message itself. It looks like you have set up mqtt broker to handle some messages but not this particular one (sensor1).

How to rollback to 2.2.1(0) depends on what your hardware platform is. You didn't indicate what OpenSprinkler hardware you have.

webdandy, 2025-03-31:
> This seems to be due to how your mqtt broker was set up. I don't see anything obviously wrong with the mqtt message itself. It looks like you have set up mqtt broker to handle some messages but not [...]

The strange thing is, the mqtt messages with 2.2.1(0) were working.
I‘m running OSPI on my Raspberry and I tried rollback to 2.2.1(0) with git, but not successfully.

rayshobby, 2025-03-31:
Could you be more specific how you tried to roll back and what error messages you got?

webdandy, 2025-03-31:
> Could you be more specific how you tried to roll back and what error messages you got?

I was searching the hash in the git log and tried the following:

root@OpenSprinkler-Pi ~/OpenSprinkler-Firmware # git checkout 1fc4c5d0a223ccc5e8f39305b57bee67c791d4fd
Hinweis: Wechsle zu '1fc4c5d0a223ccc5e8f39305b57bee67c791d4fd'.

Sie befinden sich im Zustand eines 'losgelösten HEAD'. Sie können sich
umschauen, experimentelle Änderungen vornehmen und diese committen, und
Sie können alle möglichen Commits, die Sie in diesem Zustand machen,
ohne Auswirkungen auf irgendeinen Branch verwerfen, indem Sie zu einem
anderen Branch wechseln.

Wenn Sie einen neuen Branch erstellen möchten, um Ihre erstellten Commits
zu behalten, können Sie das (jetzt oder später) durch Nutzung von
'switch' mit der Option -c tun. Beispiel:

  git switch -c <neuer-Branchname>

Oder um diese Operation rückgängig zu machen:
  git switch -

Sie können diesen Hinweis ausschalten, indem Sie die Konfigurationsvariable
'advice.detachedHead' auf 'false' setzen.

HEAD ist jetzt bei 1fc4c5d Merge branch 'dev/221_1' of github.com:OpenSprinkler/OpenSprinkler-Firmware into dev/221_1
root@OpenSprinkler-Pi ~/OpenSprinkler-Firmware # ./build.sh ospi
Building OpenSprinkler...
Updating submodules.
Installing required libraries...
Holen:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease [15,0 kB]
OK:2 http://archive.raspberrypi.org/debian bullseye InRelease
Holen:3 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf Packages [13,2 MB]
Es wurden 13,3 MB in 8 s geholt (1.729 kB/s).                                                                                                                                                                   
Paketlisten werden gelesen… Fertig
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
gpiod ist schon die neueste Version (1.6.2-1+rpi1).
libgpiod-dev ist schon die neueste Version (1.6.2-1+rpi1).
libi2c-dev ist schon die neueste Version (4.2-1+b1).
libmosquitto-dev ist schon die neueste Version (2.0.11-1+deb11u2).
libssl-dev ist schon die neueste Version (1.1.1w-0+deb11u2+rpt1).
raspi-gpio ist schon die neueste Version (0.20191001).
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 3 nicht aktualisiert.
Compiling ospi firmware...
OpenSprinkler.cpp: In static member function ‘static unsigned char OpenSprinkler::weekday_today()’:
OpenSprinkler.cpp:1765:1: error: version control conflict marker in file
 1765 | <<<<<<< HEAD
      | ^~~~~~~
OpenSprinkler.cpp:1766:3: error: expected primary-expression before ‘==’ token
 1766 | =======
      |   ^~
OpenSprinkler.cpp:1766:5: error: expected primary-expression before ‘==’ token
 1766 | =======
      |     ^~
OpenSprinkler.cpp:1766:7: error: expected primary-expression before ‘=’ token
 1766 | =======
      |       ^
OpenSprinkler.cpp:1768:1: error: version control conflict marker in file
 1768 | >>>>>>> rfstation-update
      | ^~~~~~~
OpenSprinkler.cpp: At global scope:
OpenSprinkler.cpp:1857:1: error: version control conflict marker in file
 1857 | <<<<<<< HEAD
      | ^~~~~~~
OpenSprinkler.cpp: In function ‘void send_rfsignal(ulong, ulong)’:
OpenSprinkler.cpp:1875:5: error: ‘transmit_rfbit’ was not declared in this scope
 1875 |     transmit_rfbit(len3, len);
      |     ^~~~~~~~~~~~~~
OpenSprinkler.cpp:1877:5: error: ‘transmit_rfbit’ was not declared in this scope
 1877 |     transmit_rfbit(len, len3);
      |     ^~~~~~~~~~~~~~
OpenSprinkler.cpp:1882:3: error: ‘transmit_rfbit’ was not declared in this scope
 1882 |   transmit_rfbit(len, len31);
      |   ^~~~~~~~~~~~~~
OpenSprinkler.cpp: At global scope:
OpenSprinkler.cpp:1886:1: error: version control conflict marker in file
 1886 | =======
      | ^~~~~~~

Maybe it was the wrong hash or the wrong way?

rayshobby, 2025-03-31:
Why checkout 1fc4c5d? That commit says very clearly it's merging branch 221_1, which means it's already the start of firmware 2.2.1(1). You can find all the firmware releases and their tags here:
https://github.com/OpenSprinkler/OpenSprinkler-Firmware/releases

For example, firmware 2.2.1(0) has a tag 221(0). You can do
`git checkout "221(0)"`
That switches it to firmware 2.2.1(0). Then just re-run the `build.sh` script.

rayshobby, 2025-03-31:
In addition, I am not sure why you think this is an 'error' message -- it's just a printout that tells you your mqtt client received this message but it doesn't know what to do about it. If you don't want to receive `sensor1` message, just go to your OpenSprinkler's homepage, Edit Options -> Integration tab -> Notification Events, and uncheck any message you don't want to receive.

webdandy, 2025-03-31:
This is also not working:

root@OpenSprinkler-Pi ~/OpenSprinkler-Firmware # git checkout "221(0)"
HEAD ist jetzt bei d78964b Merge pull request #288 from OpenSprinkler/support/https
root@OpenSprinkler-Pi ~/OpenSprinkler-Firmware # ./build.shBuilding OpenSprinkler...
Updating submodules.
Installing required libraries...
OK:1 http://archive.raspberrypi.org/debian bullseye InRelease
OK:2 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Paketlisten werden gelesen… Fertig
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
libgpiod-dev ist schon die neueste Version (1.6.2-1+rpi1).
libi2c-dev ist schon die neueste Version (4.2-1+b1).
libmosquitto-dev ist schon die neueste Version (2.0.11-1+deb11u2).
libssl-dev ist schon die neueste Version (1.1.1w-0+deb11u2+rpt1).
raspi-gpio ist schon die neueste Version (0.20191001).
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.
Compiling ospi firmware...
external/TinyWebsockets/tiny_websockets_lib/src/websockets_client.cpp: In member function ‘bool websockets::WebsocketsClient::send(const char*)’:
external/TinyWebsockets/tiny_websockets_lib/src/websockets_client.cpp:352:33: error: ‘strlen’ was not declared in this scope
  352 |         return this->send(data, strlen(data));
      |                                 ^~~~~~
external/TinyWebsockets/tiny_websockets_lib/src/websockets_client.cpp:6:1: note: ‘strlen’ is defined in header ‘<cstring>’; did you forget to ‘#include <cstring>’?
    5 | #include <tiny_websockets/internals/wscrypto/crypto.hpp>
  +++ |+#include <cstring>
    6 | 
external/TinyWebsockets/tiny_websockets_lib/src/websockets_endpoint.cpp: In member function ‘bool websockets::internals::WebsocketsEndpoint::send(const char*, size_t, uint8_t, bool, bool, const char*)’:
external/TinyWebsockets/tiny_websockets_lib/src/websockets_endpoint.cpp:394:21: error: ‘memcmp’ was not declared in this scope
  394 |         if (mask && memcmp(maskingKey, __TINY_WS_INTERNAL_DEFAULT_MASK, 4) != 0) {
      |                     ^~~~~~
external/TinyWebsockets/tiny_websockets_lib/src/websockets_endpoint.cpp:2:1: note: ‘memcmp’ is defined in header ‘<cstring>’; did you forget to ‘#include <cstring>’?
    1 | #include <tiny_websockets/internals/websockets_endpoint.hpp>
  +++ |+#include <cstring>
    2 | 
external/OpenThings-Framework-Firmware-Library/LinuxLocalServer.cpp: In member function ‘virtual void OTF::LinuxLocalClient::print(const char*)’:
external/OpenThings-Framework-Firmware-Library/LinuxLocalServer.cpp:53:32: error: ‘strlen’ was not declared in this scope
   53 |   client.write((uint8_t*)data, strlen(data));
      |                                ^~~~~~
external/OpenThings-Framework-Firmware-Library/LinuxLocalServer.cpp:3:1: note: ‘strlen’ is defined in header ‘<cstring>’; did you forget to ‘#include <cstring>’?
    2 | #include "LinuxLocalServer.h"
  +++ |+#include <cstring>
    3 | 
external/OpenThings-Framework-Firmware-Library/Websocket.cpp: In member function ‘bool WebsocketClient::send(const char*, size_t, bool)’:
external/OpenThings-Framework-Firmware-Library/Websocket.cpp:189:18: error: ‘strlen’ was not declared in this scope
  189 |         length = strlen(payload);
      |                  ^~~~~~
external/OpenThings-Framework-Firmware-Library/Websocket.cpp:2:1: note: ‘strlen’ is defined in header ‘<cstring>’; did you forget to ‘#include <cstring>’?
    1 | #include "Websocket.h"
  +++ |+#include <cstring>
    2 | 
root@OpenSprinkler-Pi ~/OpenSprinkler-Firmware #

webdandy, 2025-03-31:
> In addition, I am not sure why you think this is an 'error' message -- it's just a printout that tells you your mqtt client received this message but it doesn't know what to do about it. If you [...]

I want to receive ‚sensor1‘ and if I uncheck, I dont‘t receive any message ;-(

rayshobby, 2025-03-31:
Open your `build.sh`, find the line that says
`g++ -o OpenSprinkler -DOSPI $USEGPIO -DSMTP_OPENSSL ......`
then add a
`-include string.h` after the `-std=c++14` in that line. This is related to the comment here:
https://github.com/OpenSprinkler/OpenSprinkler-Firmware/issues/337#issuecomment-2746273522

closed 2025-04-02 by rayshobby