OpenSprinkler/OpenSprinkler-Firmware #344: [FEAT] Add ability to store count or heads

starsoccer, 2025-05-03:
Often times when I am turning on for the season I will do a once over to make sure all the heads are working. Currently I store the count in the description but it would be nice if it had its own field to quickly reference

rayshobby, 2025-05-05:
What do you mean by "count" or "heads"?

starsoccer, 2025-05-05:
@rayshobby I mean how many sprinklers are actually in a zone. For instance some of my zones have 4, others 5, some 6, etc.. so Id like a way to track how many each has so when it comes time to turn on each season I can confirm they are all present and working

rayshobby, 2025-05-05:
Why not just include the count in the name of the zone, like "Front Yard (4)" or something like that?

starsoccer, 2025-05-05:
That would work too I guess, just felt like having its own field that was stored would be useful rather then having to append it to each name

rayshobby, 2025-05-05:
Description is cloud synced. If you switch to a different device, you need to log in to opensprinkler.com in the app, and confirm to use cloud data to overwrite local data, in order for the description to show up.

starsoccer, 2025-05-05:
Ah okay yeah I do not have my device syncing to the cloud so that would explain why it doesnt work. It would be helpful to add that as a note somewhere in the UI if cloud sync isnt setup. Ive used description a few times and then had it wiped when I got a new device, etc..

In any case going to close this issue

closed 2025-05-05 by starsoccer